Xilinx/Vitis-AI #343: resnet50-zcu102_zcu104-r1.3.1.tar.gz not working

ttymrz, 2021-03-19:
Hi,

I'm trying Vitis-AI v1.3.1 DPU TRD Vivado Flow with ZCU102.
The [resnet50.xmodel](https://github.com/Xilinx/Vitis-AI/blob/1.3.1/dsa/DPU-TRD/app/model/resnet50.xmodel) works, but resnet50-zcu102_zcu104-r1.3.1.tar.gz downloaded from [AI-Model-Zoo](https://github.com/Xilinx/Vitis-AI/blob/1.3.1/models/AI-Model-Zoo/model-list/cf_resnet50_imagenet_224_224_7.7G_1.3/model.yaml) doesn't work.

I'm also trying to run a DPUCZDX8G_v3_3_0 with the Vivado flow on our original mpsoc board.
But xmodels created by vai_c_tensorflow does not work. 
I think there is a problem with the Vitis AI compiler. 

Thanks.

wangxd-xlnx, 2021-03-22:
Hi @ttymrz 
Could you provide some details about your operation and error info?

qianglin-xlnx, 2021-03-22:
Hi @ttymrz 
1 For resnet50-zcu102_zcu104-r1.3.1.tar.gz model, you should update the VART to 1.3.1.  Download the VART1.3.1 from https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-1.3.1.tar.gz

2 When you compile the model with the latest docker image, it's compatible with VART1.3.1.  So you should also update the VART to 1.3.1 in your custom system.

3 The app, including model and libraries, in https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/app are based on VART1.3.0.  They can be used for testing the DPU on your system. If you want to test the models based on VART1.3.1, you can use https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 for testing. Note that install the VART1.3.1 before testing.

ttymrz, 2021-03-23:
Hi

I tried the https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 with resnet50-zcu102_zcu104-r1.3.1.tar.gz on ZCU102, and I got the correct result.
I will continue to try vitis-ai v1.3.1 and our xmodel on our custom mpsoc board.
 
Thanks

```console
root@xilinx-zcu102-2020_2:~/demo/VART/resnet50# ./resnet50  resnet50.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0128 04:25:24.063268  1120 main.cc:285] create running for subgraph: subgraph_conv1

Image : bellpeppe-994958.JPEG
top[0] prob = 0.992235  name = bell pepper
top[1] prob = 0.003158  name = acorn squash
top[2] prob = 0.001915  name = cucumber, cuke
top[3] prob = 0.000905  name = zucchini, courgette
top[4] prob = 0.000549  name = strawberry
```

ttymrz, 2021-03-24:
Hi @qianglin-xlnx 

I compiled resnet50.xmodel from  [cf_resnet50_imagenet_224_224_7.7G_1.3.zip](https://www.xilinx.com/bin/public/openDownload?filename=cf_resnet50_imagenet_224_224_7.7G_1.3.zip) with vitis-ai latest docker image.
Then I tried https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 using this xmodel.
But the result is wrong.

I can't get the same xmodel as [resnet50-zcu102_zcu104-r1.3.1.tar.gz](https://www.xilinx.com/bin/public/openDownload?filename=resnet50-zcu102_zcu104-r1.3.1.tar.gz). 
It seems that the compiler and caffemodel are different.

Our plan:
- [x]  https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 with  [resnet50-zcu102_zcu104-r1.3.1.tar.gz](https://www.xilinx.com/bin/public/openDownload?filename=resnet50-zcu102_zcu104-r1.3.1.tar.gz) on ZCU102
- [ ] resnet50 with xmodel compiled  from  [cf_resnet50_imagenet_224_224_7.7G_1.3.zip](https://www.xilinx.com/bin/public/openDownload?filename=cf_resnet50_imagenet_224_224_7.7G_1.3.zip) on ZCU102
- [ ] resnet50 on our custom mpsoc board
- [ ] our custum xmodel on custom mpsoc board

Thanks

#### resnet50.xmodel compiled from cf_resnet50_imagenet_224_224_7.7G_1.3.zip

```console
root@xilinx-zcu102-2020_2:~/demo# ./resnet50 resnet50.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0129 04:03:19.816431  1130 main.cc:285] create running for subgraph: subgraph_conv1

Image : bellpeppe-994958.JPEG
top[0] prob = 0.001000  name = toilet tissue, toilet paper, bathroom tissue
top[1] prob = 0.001000  name = ear, spike, capitulum
top[2] prob = 0.001000  name = bolete
top[3] prob = 0.001000  name = hen-of-the-woods, hen of the woods, Polyporus frondosus, Grifola frondosa
top[4] prob = 0.001000  name = earthstar

root@eiger3:~/demo # xir graph resnet50.xmodel
"libs_info" = {
        "xcompiler.1.3.1" = 066ad8b036a497060b15c1ef0150f17955de4ac7
        "xcompiler.1.3.1 : target-factory.1.3.1" = f0c020088afff0f6e786922b328042d93c3e615e
        "xcompiler.1.3.1 : xir.1.3.1" = 5811a80197cc4a25ffe7055858c6db81d6e76133
}
"files_md5sum" = {
        "/workspace/resnet50/cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel" = a9cc7fb30a35551e085fdf995356dab7
        "/workspace/resnet50/cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt" = d676c2d05b43cf29b1cfb638017aa55d
        "/workspace/weight/resnet50/resnet50_org.xmodel" = e2fbeec5021c247a48b03b6460ca23d2
}
"origin" = caffe
```

#### resnet50-zcu102_zcu104-r1.3.1.tar.gz

```console
root@eiger3:~/resnet50# xir graph resnet50.xmodel
"libs_info" = {
        "xcompiler.1.2.0" = c9b864f2bf455e101cf7c443bf524b68155b558d
        "xcompiler.1.2.0 : target-factory.1.3.0" = bf887233b2a00579270fbefa709808a195b9c1e0
        "xcompiler.1.2.0 : xir.1.3.0" = a57405ab47fbe8f07e6e022558947c2ebb285386
}
"files_md5sum" = {
        "/group/modelzoo/internal-cooperation-models/caffe/resnet50.baseline9213_ck/fix/acc/decrypted/deploy.caffemodel" = f7c0228e97bcbc8f2a2951fcedc2081c
        "/group/modelzoo/internal-cooperation-models/caffe/resnet50.baseline9213_ck/fix/acc/deploy_keep_fixed_neuron/deploy.prototxt" = d676c2d05b43cf29b1cfb638017aa55d
        "/proj/xcdhdstaff1/huizhang/workspace/workspace/sft_model_zoo_builder_cloud/build_zcu102_zcu104/resnet50_parsed.xmodel" = 6a3621b9f70ee82483353ba2ce3ff0cb
}
"origin" = caffe
```

ttymrz, 2021-03-25:
Hi @qianglin-xlnx

I also tried vitis-flow using resnet50.xmodel compiled from [cf_resnet50_imagenet_224_224_7.7G_1.3.zip](https://www.xilinx.com/bin/public/openDownload?filename=cf_resnet50_imagenet_224_224_7.7G_1.3.zip). 
It worked fine with vitis-flow. But It does not work in vivado-flow.

Our application requires vivado-flow.

Thanks

ttymrz, 2021-03-25:
vitis-ai v1.3.1 and v1.3.2

- https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 with [resnet50-zcu102_zcu104-r1.3.1.tar.gz](https://www.xilinx.com/bin/public/openDownload?filename=resnet50-zcu102_zcu104-r1.3.1.tar.gz) on ZCU102
vivado-flow: OK
vitis-flow: OK

- https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 with xmodel compiled from [cf_resnet50_imagenet_224_224_7.7G_1.3.zip](https://www.xilinx.com/bin/public/openDownload?filename=cf_resnet50_imagenet_224_224_7.7G_1.3.zip) on ZCU102
vivado-flow: **NG**
vitis-flow: OK

guohaot-Xlnx, 2022-02-23:
Hi,  @ttymrz,
This issue is about to close. If you have any questions, please repost to the issues.
Regards

closed 2022-02-23 by guohaot-Xlnx