Xilinx/Vitis-AI #345: DPU freq root@xilinx - command not found

erdnag, 2021-03-22:
Hi! So I've tried the VART examples, but not all of them work. I've seen I'm not the only one having this issue and I should change the DPU frequency, but when I try to using root@xilinx... command as the others used and solved their problem, I get : command not found. In my ZCU104 terminal it's sh, not root, is this ok? It's my first time working with FPGA

qianglin-xlnx, 2021-03-24:
Hi @erdnag 
Do you use the official ZCU104 1.3.1 image? If yes, there is dpu_clk command inside. You can run the following command to check the frequency of DPU.

root@xilinx-zcu104-2020_2:~# dpu_clk
Real PL0_CLK 100000000
DPU Performance 100.0%

erdnag, 2021-03-24:
I've dowloaded it from the user guide - v1.3 and there is the ZCU104 1.3.0 image. Where can I find the 1.3.1 version? Thank you for your help!

erdnag, 2021-03-24:
I found it online, I hope it will work with this one. I'll have to try and let you know.

qianglin-xlnx, 2022-03-01:
Hi @erdnag 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-01 by qianglin-xlnx