Xilinx/Vitis-AI #347: Starting Vitis-AI docker

nevilshah235, 2021-03-24:
docker: Error response from daemon: error while creating mount source path '/opt/xilinx/dsa': mkdir /opt/xilinx: read-only file system.

hanxue, 2021-03-26:
Hi @nevilshah235 ,

Can you check that the user account that you use to run Vitis AI docker has write permissions to `/opt/xilinx` ?

nevilshah235, 2021-03-26:
Hi @hanxue

Thank you for the support. I tried running it with sudo as well. But still I get the same error.
[image]

hanxue, 2021-03-26:
Hi @nevilshah235 ,

Hmm, the error seems to be no write permission to /opt/xilinx, or /opt. Could you do me a favour and paste the output of the following commands

```
df -hTl
ls -ld /opt/xilinx
ls -ld /opt
``` 

?

nevilshah235, 2021-03-26:
Sure.

[image]
nevil@nevil-ASUS-TUF-Gaming-A17-FA706IH-FA706IH:~$ df -hTl
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  3.6G     0  3.6G   0% /dev
tmpfs          tmpfs     744M  2.4M  742M   1% /run
/dev/nvme0n1p9 ext4      288G  142G  132G  52% /
tmpfs          tmpfs     3.7G   70M  3.6G   2% /dev/shm
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     3.7G     0  3.7G   0% /sys/fs/cgroup
/dev/loop0     squashfs  100M  100M     0 100% /snap/core/10859
/dev/loop2     squashfs   56M   56M     0 100% /snap/core18/1988
/dev/loop3     squashfs  151M  151M     0 100% /snap/okular/109
/dev/loop4     squashfs  132M  132M     0 100% /snap/docker/796
/dev/loop5     squashfs  9.2M  9.2M     0 100% /snap/canonical-livepatch/95
/dev/loop6     squashfs   65M   65M     0 100% /snap/gtk-common-themes/1514
/dev/loop1     squashfs  100M  100M     0 100% /snap/core/10908
/dev/loop7     squashfs  219M  219M     0 100% /snap/gnome-3-34-1804/66
/dev/loop9     squashfs  163M  163M     0 100% /snap/gnome-3-28-1804/145
/dev/loop8     squashfs  324M  324M     0 100% /snap/kde-frameworks-5-qt-5-15-core20/14
/dev/loop10    squashfs  165M  165M     0 100% /snap/signal-desktop/351
/dev/loop11    squashfs  146M  146M     0 100% /snap/slack/37
/dev/loop12    squashfs  426M  426M     0 100% /snap/pycharm-community/232
/dev/nvme0n1p1 vfat      256M   34M  223M  13% /boot/efi
/dev/loop13    squashfs   33M   33M     0 100% /snap/snapd/11402
/dev/loop14    squashfs  163M  163M     0 100% /snap/gitkraken/172
/dev/loop15    squashfs   62M   62M     0 100% /snap/core20/904
/dev/loop16    squashfs  425M  425M     0 100% /snap/pycharm-community/226
/dev/loop17    squashfs   52M   52M     0 100% /snap/snap-store/518
/dev/loop18    squashfs   33M   33M     0 100% /snap/snapd/11107
tmpfs          tmpfs     744M   28K  744M   1% /run/user/125
tmpfs          tmpfs     744M  4.3M  740M   1% /run/user/1000
/dev/loop19    squashfs  147M  147M     0 100% /snap/slack/38
nevil@nevil-ASUS-TUF-Gaming-A17-FA706IH-FA706IH:~$ ls -ld /opt/xilinx
drwxr-xr-x 3 root root 4096 Mar 24 16:36 /opt/xilinx
nevil@nevil-ASUS-TUF-Gaming-A17-FA706IH-FA706IH:~$ ls -ld /opt
drwxr-xr-x 5 root root 4096 Mar 25 18:46 /opt
nevil@nevil-ASUS-TUF-Gaming-A17-FA706IH-FA706IH:~$

hanxue, 2021-03-26:
The `/opt` directory looks OK. Can you please list out the contents of

```
ls -l /opt/xilinx
```

please?

Just to ensure the problem is due to this directory, temporarily delete this line in your local copy of `docker_run.sh` https://github.com/Xilinx/Vitis-AI/blob/master/docker_run.sh#L87

```
    -v /opt/xilinx/dsa:/opt/xilinx/dsa \
```

hanxue, 2021-03-26:
A quick question. Are you by any chance running Docker inside a Mac or Windows machine?

We have seen weird mount issues like these, and at the moment Vitis AI is only supported on a Linux host with Docker.

nevilshah235, 2021-03-26:
Hi @hanxue 

I also deleted line no. 88 in my local installation of docker_run.sh
    -v /opt/xilinx/overlaybins:/opt/xilinx/overlaybins \
    
After that I was able to successfully run the docker_run.sh script

I am using an Ubuntu 20.04 OS. 
[image]

nevilshah235, 2021-03-26:
Sharing the output of ls -l /opt/xilinx
I had created the directory dsa, to check whether it works after I include that directory, but same issue is there

[image]

hanxue, 2021-03-26:
Which workflow are you using? If it is Whole App Acceleration or Alveo related, you will need to have XRT installed on your host. Download it from https://www.xilinx.com/products/design-tools/vitis/xrt.html#gettingstarted

This version should work

https://www.xilinx.com/bin/public/openDownload?filename=xrt_202020.2.9.317_18.04-amd64-xrt.deb

You might want to delete /opt/xilinx/dsa that you created prior to installing XRT.

nevilshah235, 2021-03-27:
Thanks for the fix suggestion. I am unable to install XRT library on Ubuntu
20.04, due to problems with prerequisite libraries.
The packages required have different version numbers. So I had not
installed XRT earlier, still facing the same issue.

hanxue, 2021-03-30:
There are a couple of options for installing XRT for Ubuntu 20.04:

Download from https://www.xilinx.com/bin/public/openDownload?filename=xrt_202020.2.9.317_20.04-amd64-xrt.deb

Build and install XRT from source per the instructions https://xilinx.github.io/XRT/2020.1/html/build.html  . i.e. Download the XRT 2020.2 official release from https://codeload.github.com/Xilinx/XRT/tar.gz/refs/tags/202020.2.8.743 , and within the XRT source directory, there is a src/runtime_src/tools/scripts/xrtdeps.sh script that will install all the required tools.

nevilshah235, 2021-03-31:
Thanks a ton for the support. I installed XRT for Ubuntu 20.04, and now the Vitis-AI docker installation is working perfectly fine.

nevilshah235, 2021-04-02:
Hi @hanxue 

I installed XRT, but it's still not working. Earlier when I had tested, I forgot to add 
    -v /opt/xilinx/dsa:/opt/xilinx/dsa \
     -v /opt/xilinx/overlaybins:/opt/xilinx/overlaybins \
 in the docker_run.sh script, so the installation worked fine. But I am still facing the same issue as mentioned earlier in starting VITIS-AI docker.

hanxue, 2021-04-02:
Hi @nevilshah235 ,

What is the exact error you are seeing? Is it the following? 
```
docker: Error response from daemon: error while creating mount source path '/opt/xilinx/dsa': mkdir /opt/xilinx: read-only file system.
```

At this point , it is easier to verify if there is something wrong with the Docker installation

1. Trying running Vitis AI docker on another machine, even if there is no FPGA and XRT installed
2. Reinstall Docker completely on your complete machine
  - back up any Docker images, if you need to preserve them
  - uninstall Docker - `sudo apt-get remove docker docker.io docker-cli containerd containerd.io docker-engine runc`
  - Remove all local docker files - `sudo rm -Rf /var/lib/docker`
  - Reinstall docker again

hanxue, 2021-04-02:
I did some googling and and notice this issue is quite common if Docker is installed via snap. 

How did you install Docker?

This should fix the snap installation

```
snap remove docker
```

Then follow the official instructions for installing Docker https://docs.docker.com/engine/install/ubuntu/

nevilshah235, 2021-04-03:
Thanks @hanxue. The issue was with the docker installation, once I reinstalled docker, it worked out fine.

closed 2021-04-19 by hanxue