Xilinx/Vitis-AI #349: Vitis AI Quantizer for custom Caffe model error

JerrySciangula, 2021-03-25:
Hi everyone,
I'm a master graduating student at Scuola Superiore Sant'Anna in Pisa.
For my thesis, I'm trying to accelerate a **custom network model** on a **ZCU102 ultrascale+** using the latest version of **_Vitis-AI tools v1.3.1._**

I installed the **Vitis AI docker for the GPU** and when I launch the following command:

**_vai_q_caffe quantize -model model_name.prototxt -weights model_name.caffemodel_**, 

the quantizer arises the error, shown in the following screenshot:

[image]

I'm using Ubuntu 20.04, with: 

**_NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2_**

The same network model has been successfully quantized and compiled on the GPU docker version for the Vitis AI v1.2.1.

What is happening now? Is there anyone that can help me?
Thanks in advance.

hanxue, 2021-03-30:
Hi Jerry,

Could you try these 2 steps:
1. Run `nvidia-smi` and make sure GPU is detected from within the Vitis AI Docker container
2. Temporarily downgrade NVIDIA driver to 411.x

qianglin-xlnx, 2022-03-03:
Hi @JerrySciangula 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx