Xilinx/Vitis-AI #351: No CUDA runtime is found Vitis AI Docker

DaniilNelubin, 2021-03-29:
Hello everyone I have been using Vitis-AI 1.3.1 docker with GPU. I'm trying to quantize my own neural network And I got the following warning and error. How can I solve this? 
```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[NNDCT_NOTE]: Loading NNDCT kernels...

Traceback (most recent call last):
  File "tool/model_quantazition.py", line 124, in <module>
    torch.cuda.set_device(args.local_rank)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/cuda/__init__.py", line 292, in set_device
    torch._C._cuda_setDevice(device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/cuda/__init__.py", line 196, in _lazy_init
    _check_driver()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/cuda/__init__.py", line 101, in _check_driver
    http://www.nvidia.com/Download/index.aspx""")
AssertionError: 
Found no NVIDIA driver on your system. Please check that you
have an NVIDIA GPU and installed a driver from
http://www.nvidia.com/Download/index.aspx
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/opt/vitis_ai/conda/envs/vitis-ai-pytorch/bin/python', '-u', 'tool/model_quantazition.py', '--local_rank=0', '--config=config/test/default_config_360_ss.yaml', '--fast_finetune', '--quant_mode=calib', '--deploy', 'dataset', 'voc2012', 'model_path', '/opt/project/init_360_torch12.pth']' returned non-zero exit status 1.
```

Eagle223, 2021-03-30:
You may need to run https://github.com/Xilinx/Vitis-AI/blob/master/setup/docker/docker_build_gpu.sh to build the Vitis-AI-gpu docker image

DaniilNelubin, 2021-03-30:
Yeah, I built this `.sh` file and add some changes in DockerfileGPU. I added COPY command and `pip install` to copy and install mseg library. Also, I changed chmod on 777 in `/opt/vitis_ai/conda/envs/vitis-ai-pytorch/` and set 777 to ALL chmod (I tried to fix permission denied error in pip install)

DockerfileGPU
```
FROM nvidia/cuda:10.0-cudnn7-runtime-ubuntu18.04
env DEBIAN_FRONTEND=noninteractive
SHELL ["/bin/bash", "-c"]
ENV TZ=America/Denver
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
ENV VAI_ROOT=/opt/vitis_ai
ENV VAI_HOME=/vitis_ai_home
ARG VERSION
ENV VERSION=$VERSION
ARG DATE
ENV DATE=$DATE
ENV LANG en_US.UTF-8
ENV LANGUAGE en_US:en
ENV LC_ALL en_US.UTF-8

RUN chmod 777 /tmp \
    && mkdir /scratch \
    && chmod 777 /scratch \
    && apt-get update -y \
    && apt-get install -y --no-install-recommends \
        apt-transport-https \
        autoconf \
        automake \
        bc \
        build-essential \
        bzip2 \
        ca-certificates \
        curl \
        g++ \
        git \
        gnupg \
        libboost-all-dev \
        libgflags-dev \
        libgoogle-glog-dev \
        libgtest-dev \
        libjson-c-dev \
        libssl-dev \
        libtool \
        libunwind-dev \
        locales \
        make \
        openssh-client \
        openssl \
        python3 \
        python3-dev \
        python3-minimal \
        python3-numpy \
        python3-pip \
        python3-setuptools \
        python3-venv \
        software-properties-common \
        sudo \
        tree \
        unzip \
        vim \
        wget \
        yasm \
        zstd

RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen \
    && echo "LC_ALL=en_US.UTF-8" >> /etc/environment \
    && echo "LANG=en_US.UTF-8" > /etc/locale.conf \
    && locale-gen en_US.UTF-8 \
    && localedef -i en_US -c -f UTF-8 -A /usr/share/locale/locale.alias en_US.UTF-8 \
    && dpkg-reconfigure locales

[... 213 lines cut ...]
    && wget -O libbutler-base.deb https://www.xilinx.com/bin/public/openDownload?filename=libbutler-base_1.3.0_amd64.deb \
    && wget -O librt-engine.deb https://www.xilinx.com/bin/public/openDownload?filename=librt-engine_1.3.0-r106_amd64.deb \
    && wget -O aks.deb https://www.xilinx.com/bin/public/openDownload?filename=aks_1.3.0-r43_amd64.deb \
    && apt-get install -y --no-install-recommends /tmp/*.deb \
    && rm -rf /tmp/* \
    && ldconfig

RUN apt-get clean -y \
    && rm -rf /var/lib/apt/lists/* \
    && rm -rf /scratch/*

# Set default build toolchain to GCC 9 for better performance
RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 90 \
        --slave /usr/bin/g++ g++ /usr/bin/g++-9 \
        --slave /usr/bin/gcov gcov /usr/bin/gcov-9 \
    && update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 80 \
        --slave /usr/bin/g++ g++ /usr/bin/g++-8 \
        --slave /usr/bin/gcov gcov /usr/bin/gcov-8 \
    && update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 70 \
        --slave /usr/bin/g++ g++ /usr/bin/g++-7 \
        --slave /usr/bin/gcov gcov /usr/bin/gcov-7

ADD docker/banner.sh /etc/

```

Almidi, 2021-09-22:
I seem to have the exact problem.
Did you find a solution ?

janifer112x, 2022-09-26:
Hi @Eagle223,
Could you please help try with our latest release(VAI 2.5)? and feel free to let me know if you still have problems.

qianglin-xlnx, 2023-04-22:
Closing since no activity for more than half year, please open a new issue if you still have any questions. Thanks.

closed 2023-04-22 by qianglin-xlnx