Xilinx/Vitis-AI #353: Vitis AI Compiler XNNC v1.3.1 error

JerrySciangula, 2021-03-29:
Hi everyone,
I'm a master graduating student at Scuola Superiore Sant'Anna in Pisa.
For my thesis, I'm trying to accelerate a **_custom network model_** on a **_ZCU102 ultrascale+_** using the latest version of **_Vitis-AI tools v1.3.1_**.

I can successfully quantize the float network model, but when I try to compile it, the following error is arised:
[image]

What does the error mean? Can anyone help me to understand what is happening?

The same model has been successfully compiled with the VAI Compiler v1.2.1. What has been changed from the two versions of the compiler?

qianglin-xlnx, 2021-03-31:
Hi @JerrySciangula 
For edge, VAI1.3 uses Xcompiler while VAI1.2 uses DNNC. 
When you quantize the model with VAI1.3, please use the option -keep_fixed_neuron.

JerrySciangula, 2021-03-31:
Hi @qianglin-xlnx, 

Can I use the -keep_fixed_option even if my hardware target platform is DPUCZDX8G and not the DPUCAHX8H, as suggested in the user guide https://www.xilinx.com/html_docs/vitis_ai/1_3/caffe.html#ariaid-title5?

jimheaton, 2021-03-31:
@JerrySciangula 
Yes, you need to use -keep_fixed_option for DPUCZDX8G when using Vitis-AI 1.3

JerrySciangula, 2021-04-01:
Hi @jimheaton,

thank you all very much! The model has been successfully compiled. 

I have one more question about the **-keep_fixed_neuron** option. What does it means? What kind of effects it has on the quantization?

jimheaton, 2021-04-01:
Hi @JerrySciangula,
New in Vitis-AI 1.3 is the xcompiler for DPUCZDX8G which use XIR (Xilinx Intermediate Representation) for the graph. The -keep_fixed_neuron option is simply needed to be compatible with XIR. It does not effect the actual quantization.

JerrySciangula, 2021-04-06:
Hi @jimheaton, thank you very much fort the reply. I solved the problem using the option -keep_fixed_neuron.

closed 2021-04-06 by JerrySciangula