Xilinx/Vitis-AI #354: Vitis AI Profiler environment problem in platform setting

HSqure, 2021-03-30:
Hi, there is a confusion about **Getting Started with Vitis AI Profiler** in `Vitis AI User Guide, UG1414 (v1.3) December 17, 2020, (page 123)`

### Document Screenshot

----------------------------------------------------------------------------------------------------------
[image]
----------------------------------------------------------------------------------------------------------

### Question description
According part **Installing the Vitis AI Profiler** in document, I finished: 
**Step a**:
[image]
And **Step b**:
[image]

**BUT** at **Step c**, I can't find option `[*] packagegroup-petalinux-self-hosted` when I run `petalinux-config -c rootfs`:
[image]

I try to ignore that **step c**, finally I got this running result with demo adas_detection when I using `vaitrace` command:

tianfang-fafafa, 2021-03-30:
hi @HSqure ,  you can try "Petalinux Package Groups  --->  packagegroup-petalinux-self-hosted   --->  [*] packagegroup-petalinux-self-hosted " , please

tianfang-fafafa, 2021-03-30:
[image]

HSqure, 2021-03-31:
> hi @HSqure , you can try "Petalinux Package Groups ---> packagegroup-petalinux-self-hosted ---> [*] packagegroup-petalinux-self-hosted " , please

Thanks for your answer! I will try it.

HSqure, 2021-04-01:
> hi @HSqure ,  you can try "Petalinux Package Groups  --->  packagegroup-petalinux-self-hosted   --->  [*] packagegroup-petalinux-self-hosted " , please

Hi @tianfang-fafafa  I tried to configure it as document and your suggestion, but same error shows as before:

sh: nm: command not found
ERROR:root:[nm --version] not exists, please check the document

tianfang-fafafa, 2021-04-01:
hi，could you please share me more information？  which platform do you use?

HSqure, 2021-04-01:
> hi，could you please share me more information？ which platform do you use?

Thanks for your reply!

The platform file I made is for `Zynq UltraScale+ MPSOC XCU15EG`. That application can fully functioning as long as I don't use `vaitrace`. 

Are there any clue about missing `GNU nm` package?

tianfang-fafafa, 2022-02-24:
hi @HSqure , we have ever resolved this issue via email. can we close this issue ?

HSqure, 2022-02-24:
> hi @HSqure , we have ever resolved this issue via email. can we close this issue ?

OK.

closed 2022-02-24 by HSqure