Xilinx/Vitis-AI #355: vai_c_tensorflow2 compile error, if layer name contains '/'

three-cups, 2021-03-30:
Hi

I'm trying Vitis-AI 1.3.2 (34794f6) and docker `xilinx/vitis-ai-gpu`, encounting error in vai_c_tensorflow2

I'm using very simple model, to test docker environment.
The network contains only one layer `keras.layers.Conv2D`.
If layer name = 'conv_conv', vai_c_tensorflow goes on.
But layer name = 'conv/conv', vai_c_tensorflow falls error.

(It's okay in TensorFlow1.x `vai_c_tensorflow`)

this is reported error message.
```
(vitis-ai-tensorflow2) Vitis-AI /workspace > vai_c_tensorflow2 -m ./quantized_vsm.h5 -a /opt/vitis_ai/compiler/arch/DPUCAHX8H/U50/arch.json -o temp -n vsm --options '{"input_shape": "1,224,224,3"}'
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=['1,224,224,3'], layout='NHWC', model_files=['./quantized_vsm.h5'], model_type='tensorflow2', out_filename='temp/vsm_org.xmodel', proto=None)
in_shapes: [[1, 224, 224, 3]]
[INFO] tensorflow2 model: quantized_vsm.h5
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 89, in to_xmodel
    model_fmt, (model_name, raw_nodes) = cls.load_raw_model(model_files)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 127, in load_raw_model
    return "h5", cls.__load_from_hdf5(model_files)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 1802, in __load_from_hdf5
    for x in param.attrs.get("weight_names").tolist()
AttributeError: 'NoneType' object has no attribute 'tolist'
```

qianglin-xlnx, 2021-04-07:
Hi @three-cups 
So far as I know, there is no limitation for Xcompiler about the layer name. Xcompiler will find the weight according to the layer's name. And the error shows that Xcompiler can't find the weight according to the layer's name. It seems that you change only the layer name without synchronously changing the mapping link between layer and weights.
A quick test, you can change the layer name from conv_conv to conv_conv_test to see if the compiler works.

three-cups, 2021-04-08:
Hi, @qianglin-xlnx 
Thank you for your replying, but I can't get result you mentioned.
My test sequence is...
1st : set layer name "conv_conv" ... success
2nd : set layer name "conv/conv" ... fail
3rd : set layer name "conv_conv_test" ... success

It may the matter of vitis_quantizer, not of Xcompiler.
Please check attached my model file (not quantized) and quantize script.
[vsm_test.zip](https://github.com/Xilinx/Vitis-AI/files/6276781/vsm_test.zip)

qianglin-xlnx, 2021-04-09:
Hi @three-cups 
Thank you for the files you provided. They are helpful and we have confirmed that it's a bug. We will fix this bug in the next release. 
Before we fix it, please avoid change the layer name with "/" . Thank you very much.

closed 2022-03-03 by qianglin-xlnx