Xilinx/Vitis-AI #356: Vitis-AI 1.3.2 Dockerfile

DaniilNelubin, 2021-03-31:
Hello, every one, as I understand you didn't update conda chanel in Dockerfile. So I can't use your fixed PyTorch quantizer in your docker container.

hanxue, 2021-04-06:
Hi @DaniilNelubin ,

The pytorch conda package are in version 1.3.1. You can check by activating the **vitis-ai-pytorch** environment

```
Vitis-AI /workspace > conda activate vitis-ai-pytorch
(vitis-ai-pytorch) Vitis-AI /workspace > conda list | grep pytorch
# packages in environment at /opt/vitis_ai/conda/envs/vitis-ai-pytorch:
pytorch                   1.4.0           py3.6_cuda10.1.243_cudnn7.6.3_0    pytorch
pytorch_nndct_cpu         1.3.1                  c22c6b35    http://artifactory/artifactory/vitis-ai-release
```

The pytorch 1.3.1 source code update itself was published in the 1.3.2 tag/release. Hope that clarifies.

DaniilNelubin, 2021-04-06:
Okay, thanks a lot for your answer, I just got confused about the versions.

closed 2021-04-06 by DaniilNelubin