Xilinx/Vitis-AI #360: [PointPillars] ZCU102 goes dead with self-compiled model

Cliu2, 2021-04-01:
Hi,

I'm trying to retrain the pointpillars model and deploy it on the ZCU102 board. But the board OS no longer response everytime when I try to run the `.cpp` demo code.

For training, I followed the codes in https://www.xilinx.com/bin/public/openDownload?filename=pt_pointpillars_kitti_12000_100_10.8G_1.3.zip, from the model-zoo https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/model-list/pt_pointpillars_kitti_12000_100_10.8G_1.3/model.yaml.

And the `.cpp` demo code I use: https://github.com/Xilinx/Vitis-AI/tree/master/demo/Vitis-AI-Library/samples/pointpillars.

For compile, I followed the command on Pg.86 of the user guide: https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf

I also tried to compile the official provided quantized model and deploy them on the ZCU102, but it goes dead as well.

Here is procedure to reproduce the problem:
1. Download the official code from https://www.xilinx.com/bin/public/openDownload?filename=pt_pointpillars_kitti_12000_100_10.8G_1.3.zip.

2. Compile the two quantized models given in the above file:
>  `vai_c_xir -x /quantized/VoxelNet_0_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json -o /VNet_0
-n VNet_0`
>  `vai_c_xir -x /quantized/VoxelNet_1_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json -o /VNet_1
-n VNet_1`

3. Copy the compiled `.xmodel` files to the corresponding folder in  ZCU102's `/usr/share/vitis_ai_library/models/`. 
The `.prototxt` files are built following the issue here #300.

4. Download https://github.com/Xilinx/Vitis-AI/tree/master/demo/Vitis-AI-Library/samples/pointpillars:
Build with `sh build.sh`,
Run with `env XLNX_POINTPILLARS_PRE_MT=1 ./test_bin_pointpillars  VNet_0 VNet_1 sample_pointpillars.bin  sample_pointpillars.png`

After start execution the system goes dead, neither the keyboard nor the mouse responses anymore.

What should I do with this problem? Thank you in advance

Eagle223, 2021-04-02:
Hello，
I think， You should check whether the Vitis ai runtime version of ZCU120 is consistent with the vitis ai compiler version.
As far as I have caused, using vitis ai runtime 1.3.1 to run the model compiled by vitis ai 1.3.0 will cause the system to crash.

Cliu2, 2021-04-04:
Thank you @Eagle223. You are right I was using the 1.3.0 compiled model on the 1.3.1 runtime. After changing to the correct version, the problem has now been solved.

closed 2021-04-04 by Cliu2