Xilinx/Vitis-AI #361: tf yolov3 error: Please specify the input shape of the model. Exit!

balintmaci, 2021-04-01:
I'm trying to compile the tensorflow based yolov3 model from the zoo for the Ultra96 board using the `compile.sh` script from DPU-PYNQ. All other tf models I've tried work fine but this one returns this error and the compilation fails.
Command: `./compile.sh Ultra96 tf_yolov3_voc_416_416_65.63G_1.3`
Output: 
```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['tf_yolov3_voc_416_416_65.63G_1.3/quantized/quantize_eval_model.pb'], model_type='tensorflow', out_filename='./tf_yolov3_org.xmodel', proto=None)
[INFO] tensorflow model: tf_yolov3_voc_416_416_65.63G_1.3/quantized/quantize_eval_model.pb
[INFO] parse raw model     :  0%|                     | 0/366 [00:00<?, ?it/s]
[ERROR] Please specify the input shape of the model. Exit!
[INFO] parse raw model     :  0%|                     | 0/366 [00:00<?, ?it/s]
```
I've tried both the GPU and CPU docker images at v1.3.411 and I tried the CPU image v1.3.598 as well.

jimheaton, 2021-04-01:
@balintmaci  
If the model does not contain the input tensor shape information you will get this error.
 Add the input layer information to the compiler as an extra option, for example:
--options "{'mode':'normal','save_kernel':'', 'input_shape':'1,416,416,3'}"

MXM107, 2021-07-15:
Hello balintmaci. Do you solve this problem? I meet this problem too. Could you tell me how to solve it? thx very much

MXM107, 2021-07-15:
@balintmaci 
Hello balintmaci. I meet this problem too. Could you tell me how to solve it? thx very much

balintmaci, 2021-07-15:
Hi @MXM107 
If you look at [the compile script](https://github.com/Xilinx/DPU-PYNQ/blob/master/host/compile.sh) you can see the compiler command at the end
```
vai_c_tensorflow \
		--frozen_pb ${MODEL_UNZIP}/quantized/quantize_eval_model.pb \
		--arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/${BOARD}/arch.json \
		--output_dir . \
		--net_name tf_${MODEL}
```

Here, I believe, you can add the aforementioned line to the list of arguments
```
vai_c_tensorflow \
		--frozen_pb ${MODEL_UNZIP}/quantized/quantize_eval_model.pb \
		--arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/${BOARD}/arch.json \
		--output_dir . \
		--net_name tf_${MODEL} \
                --options "{'mode':'normal','save_kernel':'', 'input_shape':'1,416,416,3'}"
```

closed 2021-07-15 by balintmaci

MXM107, 2021-07-15:
Hi @balintmaci 
I have add this aforementioned but this error still appear
[image]
[image]

balintmaci, 2021-07-15:
Missing "\"

MXM107, 2021-07-16:
oh thx so much! It works

Mickeyyyang, 2023-02-15:
> Hi @MXM107 If you look at [the compile script](https://github.com/Xilinx/DPU-PYNQ/blob/master/host/compile.sh) you can see the compiler command at the end
>
> ```
> vai_c_tensorflow \
> 		--frozen_pb [...]

It works, thx very much!