Xilinx/Vitis-AI #362: Vitis AI Compiler XNNC v1.3.1 error #2

JerrySciangula, 2021-04-01:
Hi everyone,
I'm a master graduating student at Scuola Superiore Sant'Anna in Pisa.
For my thesis, I'm trying to accelerate a **_custom network model_** on a **_ZCU102 ultrascale+_** using the latest version of **_Vitis-AI tools v1.3.1_**.

The model has been successfully quantized with the following command: 

**_vai_q_caffe_** quantize -model deploy.prototxt -weights deploy.caffemodel -output_dir ./q_lane_detector -keep_fixed_neuron.

When I compile the model through the command:

**_vai_c_caffe_** --prototxt q_lane_detector/deploy.prototxt --caffemodel q_lane_detector/deploy.caffemodel --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json --output_dir ./compilation --net_name lane_detector

the **_[UNILOG][FATAL][XCOM_DATA_OUTRANGE][Data value is out of range!]_** error is reported: 
[image]

Can anyone hepl me to understand the error?

Thanks in advance

qianglin-xlnx, 2021-04-06:
Hi @JerrySciangula 
Thank you for your feedback.
The error shows that some model parameters are out of DPU limits. Could you please share the quantized model so that we can further analyze the problem.

JerrySciangula, 2021-04-06:
Hi @qianglin-xlnx, the quantized deploy.prototxt and deploy.caffemodel files are attached here in the zip file:
[quantized_files.zip](https://github.com/Xilinx/Vitis-AI/files/6263994/quantized_files.zip)

Please, let me know if you need some other files.

qianglin-xlnx, 2021-04-07:
Hi @JerrySciangula 
Could you please tell me which version of quantizer do you use to quanzite the model? VAI1.2 or VAI1.3 or VAI1.3.1?  Thank you.

JerrySciangula, 2021-04-07:
Hi @qianglin-xlnx, 

I'm using the **CPU Docker Image Version:  1.3.598**, with **Build Date: 2021-02-18**. 

I tried the command vai_q_caffe -version and it gave me: 
vai_q_caffe version 1.2.5
Build Label Feb  1 2021 05:05:25

qianglin-xlnx, 2021-04-08:
HI @JerrySciangula 
Could you share the float model and few calibration data? We need them to further analyze the problem. Thank you very much.

JerrySciangula, 2021-04-08:
Hi @qianglin-xlnx, thanks a lot for the patience.

The float model is too big to load it here directly on github. Therefore, I arranged a Wetransfer link with all the information you asked me. The link enables you to download a wetransfer zip file with a directory and inside the directory you can find two zip files: one for the float model and one for the calibration images.

Link: https://we.tl/t-M0W6EWdRkB

closed 2022-04-24 by qianglin-xlnx