Xilinx/Vitis-AI #364: [PointPillars] Customize model to output more than 3 classes

Cliu2, 2021-04-07:
Hi,

I'm using the pointpillars example, and I retrained my model to predict 6 instead of 3 classes. I followed this [official code](https://www.xilinx.com/bin/public/openDownload?filename=pt_pointpillars_kitti_12000_100_10.8G_1.3.zip) for retraining and this [c++ script](https://github.com/Xilinx/Vitis-AI/blob/master/demo/Vitis-AI-Library/samples/pointpillars/test_bin_pointpillars.cpp) for deploying on a ZCU102.

I checked the output of my quantized model in docker with python and it works well to predict the total 6 classes as expected, (just before `dump_xmodel()` is called). And I have also specified in the "officialcfg.protxt" file like `num_class: 6`. However, after compiling and deploying the `.xmodel` model to ZCU102, it never gives me other classes except the predefined 3 as in Kitti.

I wonder if that is due to the c++ implementation of PointPillars has limited the output label to be within 3 classes? Please look the code in the file [pointpillars_post.cpp](https://github.com/Xilinx/Vitis-AI/blob/eeafe77e2112cdd91531d5d39cd42a3e0a444ff6/tools/Vitis-AI-Library/pointpillars/src/postprocess/pointpillars_post.cpp#L454) 
`pos = cfg_num_class == 1 ? 0 : get_max3(  &dpu_data.cls_[  anchors_mask[i]*cfg_num_class ]);`

But I don't know where I can find this file and modify it in the ZCU102.

Would you please advise what can I do to let my customized model generate predictions for 6 classes instead of 3?
Thank you very much.

lishixlnx, 2021-04-07:
if you want to support 6 classes, you need modify the c++ deploy code which is bounded with the .model tightly.
the code you found is just the part need to be modified.  
You need select the max value from 6 instead of 3 elements.
You can download the full code of Vitis-AI-Library.

Cliu2, 2021-04-08:
@lishixlnx Thank you very much for the explanation.
I only have limited experience with c++, am I understanding right that I need to:
1. Download the source code of the Vitis-AI-Library
2. Modify the corresponding `.cpp` codes like `get_max3()`
3. Compile the libraries into `.so` files
4. Copy the header files into `/usr/include/vitis/ai`, and the complied `.so` files into `/usr/lib`?

Should I follow the procedure described on Pg.33 to 36 (Setting Up the Target) from the documentation [here](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1354-xilinx-ai-sdk.pdf)?

Cliu2, 2021-04-08:
Thanks @lishixlnx 's help, I think the issue has been solved for me now.

I managed to customize the Vitis-AI library by the following steps:

1. Clone this repo, find the source files at `Vitis-AI/tools/Vitis-AI-Library`

2. Modify the source code files.
For me, to enable 6-class prediction for PointPilars, I add a new function to replace the original `get_max3()`, which simply loop through the prediction scores, and returns the index of the maximum.

3. Compile the library with the command:
`./cmake.sh --clean`

4. Find the compiled library and header files in the output directory specified by the user (or by default the `build_dir_default` in _cmake.sh_)

5. Copy (overwrite) the new _.hpp_ files into `/usr/include/vitis/ai`, and the _.so_ files into `/usr/lib`. **You may want to backup the original files first.**

6. Finally, compile the demo c++ code and run.

closed 2021-04-08 by Cliu2