Xilinx/Vitis-AI #365: quantize  result problem

zhvn100, 2021-04-07:
According to vitis-ai's tutorial, i  quantized exmple model - resnet18 , when i run following order,python resent18_quant.py   --quant_mode calib --subset 200 ,the output result is as follow:
[image]
loss is 0.597766 ,top-1 and top-5 are zero , is this situation normal?
please someone can answer me , thx

qianglin-xlnx, 2021-04-09:
Hi @zhvn100 
It's abnormal. The running result is shown below.
[image]

zhvn100, 2021-04-11:
@qianglin-xlnx  hi，thank you for answering my question.  I  download the resnet18.pth  and prepare 200 pictures in imagenet.Then , according to tutorial , i run 'python resnet18_quant.py --quant_mode calib --subset_len 200 ' in vitis-ai-cpu docker . What should i do to figure out this mistake?

clement-remillieux, 2021-05-27:
HI @zhvn100 @qianglin-xlnx ,

I have the same issue, did you found the solution ?

niuxjxlnx, 2022-02-21:
Hi, @zhvn100:

There are some desription about loss and accuracy displayed in calibration stage. Please refer to
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/README.md#run-and-output-results

closed 2022-02-22 by niuxjxlnx