Xilinx/Vitis-AI #367: Vitis AI Compiler XNNC v1.3.1 error #3

JerrySciangula, 2021-04-08:
Hi everyone,
I'm a master graduating student at Scuola Superiore Sant'Anna in Pisa.
For my thesis, I'm trying to accelerate a **custom network** model on a **ZCU102 ultrascale+** using the latest version of **Vitis-AI tools v1.3.1**.

The model has been successfully quantized with the following command:

**_vai_q_caffe quantize -model edit_deploy.prototxt -weights edit_deploy.caffemodel -calib_iter 50 -output_dir ./q_traffic_light_recognition -keep_fixed_neuron_**.

When I compile the model through the command:

**_vai_c_caffe --prototxt q_traffic_light_recognition/deploy.prototxt --caffemodel q_traffic_light_recognition/deploy.caffemodel --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json --output_dir ./compilation --net_name traffic_light_recognition_**

the following assertion error is reported:
[image]

The following zip file contains the quantized model:
[quantized_traffic_light_recognition.zip](https://github.com/Xilinx/Vitis-AI/files/6280594/quantized_traffic_light_recognition.zip)

Could anyone help me to understand the error and how to fix it?

Thanks in advance

Eagle223, 2021-04-09:
Hello@JerrySciangula,
 Could you please share the quantize information or the unquantized model so that we can further analyze the problem.

JerrySciangula, 2021-04-09:
Hi @Eagle223, thank you for your reply. 
The unquantized float model is in the following zip file:
[float_model.zip](https://github.com/Xilinx/Vitis-AI/files/6285534/float_model.zip)

Let me know if you need other files.

Eagle223, 2021-04-21:
Hello,
Sorry for late reply, 
The issue was caused by the inappropriate parsing of the strides field of the pooling layer while compiling.

JerrySciangula, 2021-04-21:
Hi @Eagle223,
can I do something to solve this problem? Or it is an internal issue of the Compiler?

zhenzhen-Xilinx, 2022-03-07:
This problem can be solved, please use the latest version.

zhenzhen-Xilinx, 2022-03-15:
Hi @JerrySciangula ,
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-15 by qianglin-xlnx