Xilinx/Vitis-AI #368: tensorflow2 Compiling issue

puppy9207, 2021-04-09:
Hello! 
I even succeeded in qauntizing FSRCNN models.
I'm using U200 models

However I don't know why there was a program error.
```
(vitis-ai-tensorflow2) Vitis-AI /workspace/juwon/FSRCNN-tf2 > vai_c_tensorflow2 -m quantized_model.h5 -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U200/arch.json -o ./compile -n fsrcnn --options '{"input_shape": "1,100,100,1"}'
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=['1,100,100,1'], layout='NHWC', model_files=['quantized_model.h5'], model_type='tensorflow2', out_filename='./compile/fsrcnn_org.xmodel', proto=None)
in_shapes: [[1, 100, 100, 1]]
[INFO] tensorflow2 model: quantized_model.h5
/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py:1809: H5pyDeprecationWarning: dataset.value has been deprecated. Use dataset[()] instead.
  value = param.get(group).get(ds).value
[INFO] parse raw model     :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 17/17 [00:00<00:00, 23294.08it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 25/25 [00:00<00:00, 41363.94it/s]
[OPT] No optimization method available for xir-level optimization.
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 25/25 [00:00<00:00, 15229.86it/s]
[INFO] generate xmodel: /workspace/juwon/FSRCNN-tf2/compile/fsrcnn_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210408-224342-90"
[UNILOG][INFO] Compiling subgraph : subgraph_quant_conv2d
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error during initialization: multiple incompatible definitions with name "argv"
Aborted (core dumped)

```

In this process, the 'fsrcnn_org.xmodel' was created, but the model was probably not compile.

puppy9207, 2021-04-09:
"my_model.h5" is original model and "quantized_model.h5" is quantized model
[models.zip](https://github.com/Xilinx/Vitis-AI/files/6283244/models.zip)

paolodalberto, 2022-07-20:
Sorry for the very late reply. I have just found out this is related to what I do. 
I took a quick look at this example and I will follow up: the transpose convolution seems not working and it is a worthy one ...

paolodalberto, 2022-07-20:
Yep, found the problem but it will not be pushed soon, sorry.
to give you a feel [4,100,100,1] for such an input 
```
Type      Number              Operations          Duration            
----------------------------------------------------------------------
LOADFM    102                 40800               41.97530864197535   
LOADW     24                  88896               8.098765432098766   
LOADTH    0                   0                   0                   
SAVE      300                 360000              370.3703703703707   
CONV      226                 6581452800          2367.010309278346   
ELEW      298                 2860800             313.68421052631464  
POOL      0                   0                   0                   
----------------------------------------------------------------------
Total Time = 2442.74 us (FPS = 6550.02)
```

paolodalberto, 2022-07-20:
[image]
I did not check the binary but the computation looks very good