Xilinx/Vitis-AI #370: confidence

LinErDeng, 2021-04-12:
Hi, why are confidence from ./test_jpeg_yolov3 small?such as the value 0.02

yongshengxu-xlnx, 2021-04-12:
Hi, LinErDeng 
    You can set the confidence by modifying the configuration file. There is a prototxt file in the directory of the corresponding model. For example, there is a yolov3_voc.prototxt file in the yolov3_voc folder. Modify the conf_threshold to what you want. The default is 0.3. We also provide an accuracy test model, the name is yolov3_voc_acc, and the accuracy test confidence is 0.005. 
    For the situation you encounter, please check the confidence setting of the model you use.

yongshengxu-xlnx, 2022-02-22:
Hi @LinErDeng
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-22 by qianglin-xlnx