Xilinx/Vitis-AI #371: libxnnpp fails to link due to duplicated symbol

alfonsosanchezbeato, 2021-04-12:
libxnnpp is failing to build with gcc 10.2.1 on Ubuntu 21.04, due to `SSDOutputInfo` being defined both in `tools/Vitis-AI-Library/xnnpp/src/multitask/ssd_detector.hpp` and `tools/Vitis-AI-Library/xnnpp/src/multitask/ssd_detector.hpp`. These are the errors I see:
```
...
[ 44%] Building CXX object xnnpp/CMakeFiles/xnnpp.dir/src/retinaface/retinaface.cpp.o
[ 44%] Building CXX object xnnpp/CMakeFiles/xnnpp.dir/src/retinaface/retinaface_post.cpp.o
[ 44%] Building CXX object xnnpp/CMakeFiles/xnnpp.dir/src/retinaface/retinaface_detector.cpp.o
[ 44%] Building CXX object pointpillars/CMakeFiles/pointpillars.dir/src/postprocess/anchor.cpp.o
[ 44%] Building CXX object xnnpp/CMakeFiles/xnnpp.dir/src/retinaface/anchor.cpp.o
[ 45%] Building C object xnnpp/CMakeFiles/xnnpp.dir/version.c.o
[ 45%] Building CXX object pointpillars/CMakeFiles/pointpillars.dir/src/postprocess/pointpillars_post.cpp.o
[ 45%] Building CXX object pointpillars/CMakeFiles/pointpillars.dir/src/postprocess/parse_display_result.cpp.o
[ 45%] Linking CXX shared library libxnnpp-xnnpp.so
./tools/Vitis-AI-Library/xnnpp/src/multitask/././ssd_detector.hpp:30:8: error: type ‘struct SSDOutputInfo’ violates the C++ One Definition Rule [-Werror=odr]
./tools/Vitis-AI-Library/xnnpp/src/ssd/././ssd_detector.hpp:28:8: note: a different type is defined in another translation unit
./tools/Vitis-AI-Library/xnnpp/src/multitask/././ssd_detector.hpp:31:12: note: the first difference of corresponding definitions is field ‘order’
./tools/Vitis-AI-Library/xnnpp/src/ssd/././ssd_detector.hpp:29:10: note: a field with different name is defined in another translation unit
./tools/Vitis-AI-Library/xnnpp/src/multitask/././ssd_detector.hpp:30:8: note: type ‘struct SSDOutputInfo’ itself violates the C++ One Definition Rule
./tools/Vitis-AI-Library/xnnpp/src/ssd/././ssd_detector.hpp:28:8: note: the incompatible type is defined here
./tools/Vitis-AI-Library/xnnpp/src/multitask/././ssd_detector.hpp:30:8: note: type ‘struct SSDOutputInfo’ itself violates the C++ One Definition Rule
./tools/Vitis-AI-Library/xnnpp/src/ssd/././ssd_detector.hpp:28:8: note: the incompatible type is defined here
```

qianglin-xlnx, 2021-04-20:
Hi @alfonsosanchezbeato 
Thank you for your feedback. It's a bug triggered by gcc10.2.1. We will fix it in VAI1.4.

levent9402, 2022-02-17:
Hi，Do you still have this issue after using the latest version?

guohaot-Xlnx, 2022-02-23:
Hi @alfonsosanchezbeato,
This issue is about to close. If you have any questions, please repost to the issues

closed 2022-02-23 by guohaot-Xlnx

alfonsosanchezbeato, 2022-03-18:
Thanks, I can confirm now that this is not an issue anymore.