Xilinx/Vitis-AI #373: How to proceed after .elf file is no more?

kevinkit, 2021-04-14:
Hello,

I am using a ZCU106 board, so I am using a custom platform flow. I am using the VITIS-flow, and now with VITIS-AI 1.3 some changes came, such as that the .elf file is no more. 

Usually my workflow was the following:

1. Start Docker (gpu-latest)
2. Use the vitis-ai-tensorflow conda env
3. Run dlet with my system.hwh
4. Adapt the arch.json according to my dcf file / DPU architecture
5. Compile it using vai_c_tensorflow
6. Use the created .elf file and copy it into my src folder of the VITIS DPU Application 
7. Include the .elf file via GCC Host Linker Settings
8. Build project

This led to creation of the sd_card folder withing the DPU Application, but now I am missing the .elf file due to the change to the xmodel, how can I proceed after step 6 ? 

Thank you for your help

qianglin-xlnx, 2021-04-27:
Hi @kevinkit 
The .elf file is replaced by xmodel file. There is no need to include xmodel file via GCC host linker setting. 
After you compile the model into xmodel file, you need to copy the xmodel file to your target. Then build your application. The xmodel file will be used as an input parameter to the application.
You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50 
Note that you need to install VART before you run the program.

qianglin-xlnx, 2022-02-22:
Hi @kevinkit 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-22 by qianglin-xlnx