Xilinx/Vitis-AI #375: HDL representation Bitstream

ChristiaanBoe, 2021-04-14:
Hello,

I wish to use Vitis-AI to run a YOLOv4 model on an ultra96v2 model.

Is it possible to find the HDL(such as verilog) represenatiation somewhere of the generated bitstream or does Vitis-AI not support this?

Thank you in advance

qianglin-xlnx, 2021-04-20:
Hi @ChristiaanBoe 
Vitis-AI supports Ultra96-v2. Find more details in https://www.hackster.io/AlbertaBeef/vitis-ai-1-3-flow-for-avnet-vitis-platforms-cd0c51
For HDL of DPU, it's not open source. It's encrypted. You can integrate DPU with Vitis flow or Vivado flow.
For Vitis flow, you can refer to https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/prj/Vitis
For Vivado flow, you can refer to https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/prj/Vivado

ChristiaanBoe, 2021-05-25:
Dear @qianglin-xlnx 

Thank you for your concrete reply, I presume then that it also not possible to see the netlist of the generated DPU/

Kind regards,

Christiaan Boerkamp

JasonXijieJia, 2022-02-21:
Hi @ChristiaanBoe 
 Sure, the netlist of DPU is also encrypted.

closed 2022-02-25 by qianglin-xlnx