Xilinx/Vitis-AI #377: CLIENT_CONNECTION_ERROR in Speaking to Butler

rubende, 2021-04-15:
Hello,

We are having problems with docker version 1.3.1 to communicate with the FPGA (U200). This was not happening with the previous version of docker, so we assume it is related to the version change. 

We are following [this tutorial](https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADX8G/tensorflow) and, in the section "Partition, Compile and Run Inference", it gives the following error:

```
Speaking to Butler 
client connection error
Response from Butler is: 
errCode: errCode: 1
errCode String: CLIENT_CONNECTION_ERROR
myHandle: 0
valid: 1
```

Command `xbutil validate`, that works correctly outside of docker, gives the following error into the docker:

```
EXE: /opt/xilinx/xrt/bin/unwrapped/xbutil
[XRT] ERROR: unable to reserve virtual CU
ERROR: Failed to download xclbin: verify.xclbin
ERROR: == verify kernel test FAILED
INFO: Card[0] failed to validate.
```

Also, command `/opt/xilinx/xrt/bin/unwrapped/xbutil2 validate` give a similar error:
[image]

As the warning indicates, we are working with a version of CentOS previous to the recommended one, but in the previous version of docker this also happened and did not generate communication problems with the FPGA.

Any suggestions? 
Maybe we are missing some configuration step?

Thanks.

hanxue, 2021-04-27:
Seems like compiling the xclbin with the currently installed kernel on your machine, **3.10.0** causes error. Per the earlier warning message, XRT officially supports kernel 5.3.0 and above. 

Can you try upgrading the CentOS to kernel 5.x using ELRepo? 
https://elrepo.org/tiki/HomePage

The steps commands using **sudo/root**
```
rpm --import https://www.elrepo.org/RPM-GPG-KEY-elrepo.org
yum install https://www.elrepo.org/elrepo-release-7.el7.elrepo.noarch.rpm
yum --enablerepo=elrepo-kernel install kernel-lt kernel-lt-headers kernel-lt-tools kernel-lt-tools-libs
```

Check that GRUB will boot up with the new kernel, and reboot. Making sure that the new 5.x kernel is in use. 

```
$ uname -a
Linux xcosda142 5.4.114-1.el7.elrepo.x86_64 #1 SMP Mon Apr 19 18:53:45 EDT 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Then try reinstalling XRT

```
sudo yum install -y ./xrt_202020.2.8.726_7.4.1708-x86_64-xrt.rpm
```

rubende, 2021-05-04:
Reinstalling xButler has solved the problem. Thank you.

closed 2021-05-04 by rubende