Xilinx/Vitis-AI #379: Implementing VART ADAS example in Python

peterdors, 2021-04-19:
Hello, I would like to try and use the Python API to reimplement the current ADAS example in C++ for VART. Would this be possible to implement with the current Python API? The only examples I find in VART using Python API involve image classification so I am unsure of how to complete this. 

Thanks!

qianglin-xlnx, 2021-04-20:
Yes, you can implement VART ADAS example with Python API。

peterdors, 2021-04-21:
Okay thank you for the confirmation. Are there any examples to come out with newer releases of Vitis AI that would have these implemented in Python?

qianglin-xlnx, 2021-04-23:
Hi @peterdors 
You can refer to https://github.com/Xilinx/Vitis-AI/blob/master/demo/VART/resnet50_mt_py/resnet50.py
For the pre-process and post-process fucntion, users need to implement these functions themselves.

closed 2021-05-25 by peterdors