Xilinx/Vitis-AI #381: vitis-ai debug env

simplelins, 2021-04-19:
Hi, 
   I want to build a vitis-ai debug env to run the a resnet50 model with gdb. I checked out the branch '1.3.0', and then build the libs at folder "Vitis-AI-Library" and "Vitis-AI-Runtime". when I use the debug lib to run the the model resnet50 at path"Vitis-AI/examples/DPUCADF8H/tf_resnet" it will cause Segmentation fault.as below:

[image]

I have four question:

1. by gdb I can find that the error for dlopen the lib "librt-engine.so",  I donot know the reason ,pleae help,
[image]
2.where to get the source of the lib "librt-engine.so"? can I build the lib "librt-engine.so" by myself?
3.when compling the resnet50 model, I find that will use the tool "xcomplier" or "xfdnn compiler", Is this tool open source?
4.can I use the DPUCADF8H as a IP beacause I want to encrypt/decrypt the data at fpga?  if so please tell how to do. I tried to send e-mail to  'xilinx_ai_prod_mkt@xilinx.com', but send failed.

thanks

aaronn-xilinx, 2021-04-21:
Hi @simplelins could you please tell us where you got your resnet50 .xmodel from? 
The DPUCADF8H runner depends on attributes in the .xmodel to determine things like the tensor shape. And it looks like your .xmodel does not have those attributes. Your .xmodel might be an older version.
Could you please try generating a new .xmodel using the latest xcompiler?

simplelins, 2021-04-22:
hi， where to get the latest xcompiler?  the .xmodel I used came from compiling by myself 
 as "examples/DPUCADF8H/tf_resnet/README.md". 
I had tried to using the xmodel that came from 'https://www.xilinx.com/bin/public/openDownload?filename=resnet50-u200-u250-r1.3.0.tar.gz'  successfully.

the question four is important for me ,please help, thanks

aaronn-xilinx, 2021-04-27:
We will release instructions to use DPUCADF8H as an IP either in Vitis 1.5 or 1.6. It is a work in progress.

SuZhan0322, 2022-09-01:
I think this issue have been resolved in the latest vai release. Please reopen if you have any other questions about this issue. thanks!

closed 2022-09-01 by SuZhan0322