Xilinx/Vitis-AI #384: Quantify the PyTorch model

dingguodong-826, 2021-04-20:
Hello, I had this problem when I was quantifying the trained model with version 1.3, except that I didn't print any information.
[image]
It should be noted that there are two parts of my network that have reshape operation, but after I load the pre-trained weights, the network can print out the output information.Thank you very much!Good luck with your work.

dingguodong-826, 2021-04-20:
If you like, I can send you my code via email

carrotandjzy, 2021-04-27:
It' is caused by input ndim equal with output ndim for view op. This issue will be fixed in version 1.4.

niuxjxlnx, 2022-02-21:
Suppose the issue has been fixed in Vitis-AI 1.4. Will close it if no more response.

closed 2022-02-22 by niuxjxlnx