Xilinx/Vitis-AI #385: picture

LinErDeng, 2021-04-20:
./test_jpeg_yolov3 
the input picture size is ??

yongshengxu-xlnx, 2021-04-22:
The input image of test_jpeg_yolov3 can be of any size, but the network input is fixed, and the inconsistency of the input size and the network input may affect the detection effect.
   Take yolov3_voc as an example. Its network input is 416x416. If you enter a picture of any size, it will be resized to 416x416. After the detection frame is obtained, it will be zoomed back to the position corresponding to the original picture.

LinErDeng, 2021-04-22:
if I set the input size  to 608*608 in cfg file, then generate the elf file.
Is the input of the network is 608*608?

yongshengxu-xlnx, 2021-04-22:
yes, you can change input size to 608x608 in cfg file

qianglin-xlnx, 2022-02-18:
Hi @LinErDeng 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-18 by qianglin-xlnx