Xilinx/Vitis-AI #386: pytorch model compilation issue

liuhongduosc, 2021-04-21:
when I compile a PyTorch model after quantization, the following command could work:
```
vai_c_xir -x ./mymodel.xmodel -o ./compile_result/ -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U200/arch.json
```
while the following cannot:
```
vai_c_xir -x ./mymodel.xmodel -o ./compile_result/ -a /opt/vitis_ai/compiler/arch/DPUCADX8G/ALVEO/arch.json
```
The error information  is: NO FRONT END SPECIFIED.
It seems that arch.json of DPUCADX8G cannot work.

qianglin-xlnx, 2021-04-25:
Hi @liuhongduosc 

For DPUCADX8G compiling, please refer to https://www.xilinx.com/html_docs/vitis_ai/1_3/compiling_model.html#win1576132261321
Pytorch model is not supported by DPUCADX8G

liuhongduosc, 2021-04-25:
Hi,

Thanks for your reply. Do you have any plan to support DPUCADX8G for PyTorch model in your later release?

qianglin-xlnx, 2021-04-28:
Hi @wilderfield Do you know if we have any plan to support DPUCADX8G for PyTorch model?

SuZhan0322, 2022-09-01:
I think the issue have been resolved in the latest VAI release. Please reopen the issue if you have any other questions. Thanks!

closed 2022-09-01 by SuZhan0322