Xilinx/Vitis-AI #393: xbutler service must be restarted after each reboot

boxerab, 2021-04-22:


hanxue, 2021-04-27:
@boxerab The `xbutler` service is started automatically on boot. You can run a quick check with systemctl

```
$ systemctl list-unit-files | grep xbutler | grep enabled
xbutler.service                               enabled 
```

If the FPGAs in a system are not accessible at the time xbutler starts, xbutler will think no cards exist in the system and will not run. 

Is this the situation, there are no FPGAs detected?

boxerab, 2021-04-27:
@hanxue thanks. Here is the output from `service xbutler status` after reboot:

```
● xbutler.service - The Xilinx Butler
   Loaded: loaded (/etc/systemd/system/xbutler.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-04-27 08:14:52 EDT; 1min 3s ago
 Main PID: 1471 (xbutler)
    Tasks: 3 (limit: 19660)
   CGroup: /system.slice/xbutler.service
           ├─1471 /bin/bash /etc/init.d/xbutler start
           └─1586 /usr/sbin/xbutler

Apr 27 08:14:52 caliban xbutler[1471]: UID: 0
Apr 27 08:14:52 caliban xbutler[1471]: [Tue Apr 27 12:14:52 2021 GMT]
Apr 27 08:14:52 caliban xbutler[1471]: HOST: caliban
Apr 27 08:14:52 caliban xbutler[1471]: EXE: /usr/sbin/xbutler
Apr 27 08:14:52 caliban xbutler[1471]: [XRT] ERROR: No devices found
Apr 27 08:14:52 caliban xbutler[1471]: -----------
Apr 27 08:14:52 caliban xbutler[1471]: System Info
Apr 27 08:14:52 caliban xbutler[1471]: -----------
Apr 27 08:14:52 caliban xbutler[1471]: Starting Server Thread...
Apr 27 08:14:52 caliban xbutler[1471]: Done Starting Butler!
```

And here is the status after restart:

```
● xbutler.service - The Xilinx Butler
   Loaded: loaded (/etc/systemd/system/xbutler.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-04-27 08:17:27 EDT; 1s ago
  Process: 6385 ExecStop=/etc/init.d/xbutler stop (code=exited, status=0/SUCCESS)
 Main PID: 6424 (xbutler)
    Tasks: 4 (limit: 19660)
   CGroup: /system.slice/xbutler.service
           ├─6424 /bin/bash /etc/init.d/xbutler start
           └─6488 /usr/sbin/xbutler

Apr 27 08:17:27 caliban xbutler[6424]: ------------
Apr 27 08:17:27 caliban xbutler[6424]: Reading config file from: /etc/xbutler/xbutler.config
Apr 27 08:17:27 caliban xbutler[6424]: -----------
Apr 27 08:17:27 caliban xbutler[6424]: System Info
Apr 27 08:17:27 caliban xbutler[6424]: -----------
Apr 27 08:17:27 caliban xbutler[6424]: FPGA #0: xilinx_u250_xdma_201830_2
Apr 27 08:17:27 caliban xbutler[6424]: FPGA #1: xilinx_u50_gen3x16_xdma_201920_3
Apr 27 08:17:27 caliban xbutler[6424]: -----------
Apr 27 08:17:27 caliban xbutler[6424]: Starting Server Thread...
Apr 27 08:17:27 caliban xbutler[6424]: Done Starting Butler!

```

wilderfield, 2021-04-27:
Aaron,

Thanks for sharing the log.

xbutler[1471]: [XRT] ERROR: No devices found
This indicates that you are hitting the issue that Hanxue described.

We will make sure this gets fixed heading into the next release.

FYI, we are planning to move away from “xbutler” and use the already open sourced “XRM” for resource allocation. Just letting you know so you aren’t blind sided. XRM and butler are very similar. They are both a Linux daemons that listen for requests for device/kernel allocation.

boxerab, 2021-04-27:
@wilderfield thanks, looking forward to the fix, and also to switch to XRM.

janifer112x, 2022-09-05:
closed it as we've move xbutler to XRM in new Vitis AI release

closed 2022-09-05 by janifer112x