Xilinx/Vitis-AI #395: Custom pointpillars model on ZCU104

reddittoob, 2021-04-26:
Hi, I'm trying to run custom pointpillars model on ZCU 104.

However, I have some problem when I ran on ZCU board.

My result is like below.
[image]

I thought training, quantization, and compile were all done. I don't know why this problem. 

Anyone somebody know about this?

Thank you.

lishixlnx, 2021-04-27:
please try to run the model with python code before compile it and check the result if it's ok.

reddittoob, 2021-04-28:
There is not visualization and evaluation can't run because I trained model without images and calibration. So, my evaluation results have only zero.

lishixlnx, 2021-04-28:
first please try to find out if your trained model is correct.
that can be test by python code.

reddittoob, 2021-04-29:
What python code can be used test?

I have one more question. When I create ground truth, I resized a 3d bounding box height to make correct gt. 

Is this affect to train process?

lishixlnx, 2021-04-29:
for first question:
since you have trained your model, 
does your trained model run correctly?
Only if you are sure the trained model runs well then you can continue with the quantize, compile, etc.
You need run the model each step to make sure it's correct.

reddittoob, 2021-04-29:
Thank you for your reply.

I tried to test my trained model, however, I cannot run my model because I didn't have image and calibration.

lishixlnx, 2021-04-30:
do you mean binary points file when you say "image"?
I can't understand that you can train your model but can't run your model.

reddittoob, 2021-05-03:
In pointpillars code, I need binary points, images, and calibration. I mean just "image".

However, I didn't have image and calibration. I have just binary points and labeling data.

reddittoob, 2021-05-03:
No, I mean just image used in code.

In testing code, image and calibration data are needed.

closed 2021-05-26 by reddittoob