Xilinx/Vitis-AI #397: Unable to build local GPU container

denishem, 2021-04-26:
Hi,

I am trying to build the GPU container as explained in the documentation:
```
cd setup/docker
./docker_build_gpu.sh
```
But the process stops and hangs on ``Processing triggers for libc-bin (2.27-3ubuntu1.4) ...`` and nothing happens next.

I am using Ubuntu 20.04.2 with Docker version 20.10.6.

hanxue, 2021-04-27:
There could be multiple reasons for a docker build to hang - network connection, out of disk space, lack of memory (RAM). Usually a <kbd>Ctrl</kbd> + <kbd>C</kbd> and restarting the build will solve the problem. 

When running `./docker_build_gpu.sh`, you can open another terminal and run `docker stats` to see the resource usage. 

Another method is to delete **all docker containers and images** with the command

```
$ docker system prune
```

denishem, 2021-04-27:
Internet connection is good, my machine has plenty of disk space left and 32 GB of RAM. I tried removing everything and build again but the result is the same. When the process hangs on libc-bin triggers here is what `docker stats` shows:
```
CONTAINER ID   NAME               CPU %     MEM USAGE / LIMIT     MEM %     NET I/O   BLOCK I/O        PIDS
0f7dc6bf92ce   frosty_chaplygin   0.00%     142.5MiB / 31.11GiB   0.45%     0B / 0B   14.4MB / 573MB   4
```

lebacuong, 2022-08-23:
> Problem solved, this was a proxy issue.

Hi @denishem 
I am facing same promblem with you( but i stucked at vitis-ai-tensorflow installation). Can you give me a suggesstion for this problem?
Thanks you!