Xilinx/Vitis-AI #398: Clarification Regarding E2E Latency Numbers

zohourih, 2021-04-26:
I wanted to clarify the way the E2E latency numbers are calculated and posted on this page:
https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo#performance-on-zcu104
Specifically in the case of ZCU104, there are two DPU engines instantiated and the benchmarking is done with a batch size of 2 by default based on this message:

jimheaton, 2021-04-26:
The E2E latency number is using a single thread (one DPU, batch size =1), and in this case the frame rate is inverses of the latency. 

The multithreaded throughput value uses 2 DPUS, batch size =2.

zohourih, 2021-04-26:
@jimheaton Based on the the reply from @qianglin-xlnx, that is not the case. Single-threaded is still batch size of 2 and number of threads is independent of batch size, unless the person doing the benchmarking modified the Xilinx benchmark code to enforce batch size of 1 as mentioned under that Github issue.

jimheaton, 2021-04-26:
@zohourih. Issue #275 was a question about the Alveo U50 DPU. 
For Zynq and ZynqUltraScale+ DPUs: One thread = 1 DPU, batch size 1.

qianglin-xlnx, 2021-04-27:
Hi @zohourih 
Just as @jimheaton  said, for DPU on edge, the batch size is 1. 
For ZCU104, we can integrate 2 DPU core, that batch size of each DPU core is 1. The number of DPU core is not the batch num.

zohourih, 2021-04-27:
@jimheaton @qianglin-xlnx All right, I understand your point. I will close the issue, then.

closed 2021-04-27 by zohourih