Xilinx/Vitis-AI #401: quant tf.pb with vitis-ai1.2.1

dingguodong-826, 2021-04-29:
Hello, the following error occurred when I was quantifying the tf.pb file with 1.2.1
It should be noted that I converted the PyTorch model to TF, and after the conversion, I tested the output of the TF model, which was the same as the output of the PyTorch. There is no problem with the tf.pb file;Only the quantize_eval_model.pb file is generated under the vai_q_output folde
[image]
[image]

xieyi4650, 2022-02-23:
are you set correct output node?   this error occur when fixneuron op( created by quantize tool) did not run.

qianglin-xlnx, 2022-08-24:
Hi @dingguodong-826 
Closing since no activity for more than half year, please reopen if you still have question, thanks.

closed 2022-08-24 by qianglin-xlnx