Xilinx/Vitis-AI #402: point cloud range

reddittoob, 2021-05-03:
Hi, I have problem in training pointpillars.

My point cloud range of custom dataset is different with kitti dataset. 

When I changed point cloud range in config file and run train.py, I got an error like below.

  File "/workspace/demo/Vitis-AI-Library/samples/pointpillars/pt_pointpillars_kitti_12000_100_10.8G_1.3/code/train/second/pytorch/models/voxelnet.py", line 501, in forward
    x = torch.cat([up1, up2, up3], dim=1)
RuntimeError: invalid argument 0: Sizes of tensors must match except in dimension 1. Got 128 and 125 in dimension 2 at /opt/conda/conda-bld/pytorch_1579027003190/work/aten/src/THC/generic/THCTensorMath.cu:71

How can I chaned my point cloud range?

Treemann, 2021-05-06:
Hi, @reddittoob. When you change the point cloud range and use the original backbone configuration, remember to adjust your point cloud range & voxel_size to make the height&width of the BEV pseudo-image could be divided by 8, for the reason that the tensor 'up3' is upsampled from the feature map (stride=8) of the BEV pseudo-image.
What's more, if you change the point cloud range, the anchor offset should be adjusted accordingly.

reddittoob, 2021-05-07:
Thank you for reply. 

Can you let me know how to adjust anchor offset?

Treemann, 2021-05-07:
For example, if point_cloud_range: xmin=0, ymin=-39.68 (which is the origin_offset), then **offsets = origin_offset + strides / 2**.

```
      anchor_generators: {
         anchor_generator_stride: {
           sizes: [1.6, 3.9, 1.56] # wlh 
           strides: [0.32, 0.32, 0.0] # if generate only 1 z_center, z_stride will be ignored
           **offsets: [0.16, -39.52, -1.78] # origin_offset + strides / 2** 
           rotations: [0, 1.57] # 0, pi/2
           matched_threshold : 0.6 
           unmatched_threshold : 0.45
         }   
       }   
```

reddittoob, 2021-05-07:
In the config file, there are 2 anchor_generators.

Do I have to change 2 anchor_generator?

Treemann, 2021-05-07:
If your point_cloud_range is [0, -20, -2.73, 40, 20, 1.27], then the origin_offsets of x, y are 0 and -20.
If you use anchor_generator_stride and set "strides: [0.32, 0.32, 0.0]", then according to the formula: "offsets = origin_offset + strides / 2", the offsets should be set "offsets: [0.16, -19.84, ?]", the z offset could be set according to the mean center of bboxes of your object class.

reddittoob, 2021-05-07:
Although I changed range and offset, I got an same error. 

I guess this error occurred by voxel_size. How do you think about?

Treemann, 2021-05-07:
Agree.  
Point cloud range & voxel_size should be adjusted to make the height&width of the BEV pseudo-image could be divided by 8.

reddittoob, 2021-05-07:
Is there any calculate formula?

Treemann, 2021-05-07:
If you setting is 
```
point_cloud_range: [0, -20, -2.73, 40, 20, 1.27]
voxel_size: [0.16, 0.16, 4]
```
The H&W of the BEV pseudo-image:
(40-0)/0.16=250
(20-(-20))/0.16=250
Because 250 could not be divided by 8, then error occurs.

If you want to keep the voxel_size, you could adjust the point_cloud_range slightly:
```
point_cloud_range: [0, -20.48, -2.73, 40.96, 20.48, 1.27]
voxel_size: [0.16, 0.16, 4]
```
Because (40.96-0)/0.16=256 and 256 could be divided by 8, then this error will not occur.

Both point_cloud_range&voxel_size could be adjusted to meet your performance requirements and avoid such error.

reddittoob, 2021-05-10:
Thank you for your help.

I'm sorry, but I have one more question.

When I ran bash prepare_data.sh, my ground truth results are little strange. 

When I used kitti dataset, ground truth results are good. However, when I used my custom dataset, results are like below.

[image]

For example, pedestrian ground truth results look like the head and legs are ct off.

Do you know why occurred this problem?

Treemann, 2021-05-10:
You could check if the definition of your annotation is exactly the same as KITTI's.

reddittoob, 2021-05-10:
Annotation is absolutely exact...

I already check all  annotation.

Treemann, 2021-05-10:
If so, I'm sorry for no idea about the problem of your custom dataset.

reddittoob, 2021-05-10:
Ok. Thank you for your help.

qianglin-xlnx, 2021-05-26:
hi @reddittoob 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2021-05-26 by qianglin-xlnx