Xilinx/Vitis-AI #405: cannot load symbol 'create_runner'! lib=librt-engine.so, error=/usr/lib/librt-engine.so: undefined symbol: create_runner_with_attrs

weberxzq, 2021-05-06:
I am following the instruction from https://github.com/Xilinx/Vitis-AI and want to run a demo on my Alveo U280 card on CentOS 7.5.1804 However, when I entered vitis-ai-cpu on docker and walked throught the demo example, I got this error when running this: 
./test_jpeg_facedetect densebox_320_320 sample_facedetect.jpg

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0419 21:49:05.881804 17462 dpu_runner.cpp:240] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot load symbol 'create_runner'! lib=librt-engine.so, error=/usr/lib/librt-engine.so: undefined symbol: create_runner_with_attrs
*** Check failure stack trace: ***
Aborted (core dumped)

I have been stuck on this for a month, posted on Xilinx forum and nobody can actually solve this issue. I strictly followed the instruction here, both docker and the vitis-ai images are the latest version, and I did not modify anything. 

Someone on replied on the Xilinx forum (https://forums.xilinx.com/t5/AI-and-Vitis-AI/Vitis-AI-demo-example-failed/m-p/1231996), suggesting run the command:
$ nm -D /usr/lib/librt-engine.so | grep create_runner
and expect to see this:
0000000000039820 T create_runner
00000000000397e0 T create_runner_with_attrs
However, when I ran this command, I got this: 
Vitis-AI /workspace > nm -D /usr/lib/librt-engine.so | grep create_runner
0000000000030e90 T create_runner
so no create_runner_with_attrs. 

Please help, I have been stuck here for a long time, no idea how to fix it. Thanks a lot.

wanghy-xlnx, 2021-05-07:
Hi @weberxzq ,
Please try to download and install https://www.xilinx.com/bin/public/openDownload?filename=librt-engine_1.3.0-r130_amd64.deb in your docker container, then run the demo example again.

weberxzq, 2021-05-07:
After installing librt-engine_1.3.0-r130 inside the docker, the previous issue is gone, but the another issue showed up. When I walked through the same example of densebox 320 320, after running ./test_jpeg_facedetect densebox_320_320 sample_facedetect.jpg first time, it seems to be OK with the following message showed up

Vitis-AI /workspace/demo/Vitis-AI-Library/samples/facedetect > ./test_jpeg_facedetect densebox_320_320 sample_facedetect.jpg
loading xclbin: /opt/xilinx/overlaybins/dpu.xclbin
Device acquired (New CU)
errCode: errCode: 0
errCode String: SUCCESS
myHandle: 1
valid: 1

done loading xclbin
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0507 02:19:20.549386  5795 process_result.hpp:25]  0.997199 0.19375 0.14375 0.146875 0.15625
I0507 02:19:20.549672  5795 process_result.hpp:25]  0.994089 0.528125 0.1 0.14375 0.1625

So the above looks fine. However, if I ran the same thing again or run with a different image in image/ folder, it has the following issue: 

Vitis-AI /workspace/demo/Vitis-AI-Library/samples/facedetect > ./test_jpeg_facedetect densebox_320_320 sample_facedetect.jpg
loading xclbin: /opt/xilinx/overlaybins/dpu.xclbin
Device acquired (New CU)
errCode: errCode: 0
errCode String: SUCCESS
myHandle: 2
valid: 1

done loading xclbin
LOAD START:0
LOAD END  :0
SAVE START:0
SAVE END  :0
CONV START:0
CONV END  :0
MISC START:0
MISC END  :0
IP COUNTER: 4294967295
0x80:0
0x84:0
0x88:0
0x8c:0
0x90:0
0x94:0
0x98:0
0x9c:0
0xa0:0
0xa8:4294967295

0x100:81920
0x104:1
0x108:606208
0x10c:1
0x110:0
0x114:0
0x118:0
0x11c:0
0x120:0
0x124:0
0x128:0
0x12c:0
0x130:0
0x134:0
0x138:0
0x13c:0
0x140:45056
0x144:1

0x1ec:1

0x1f0:29
0x1f4:50331648
0x1f8:0
0x1fc:0
###############################
Error: CU timeout when do preload 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error: CU timeout when do preload 0
Aborted (core dumped)

If I ran it again and again, the myHandle will increment by 1 and still show the same error message at preload. 
Any ideas? I think there are some issues with this vitis version because I just strictly followed the instruction here and did not modify anything. Thanks.

wanghy-xlnx, 2021-05-10:
Hi @weberxzq ,
Please try the newly provided model below.
[densebox_320_320.tar.gz](https://github.com/Xilinx/Vitis-AI/files/6451004/densebox_320_320.tar.gz)
Best Regards

weberxzq, 2021-05-11:
Hi @wanghy-xlnx,

I tried this new densebox_320_320.tar.gz, it is a little bit better as it worked fine for the first 3-4 images and then failed again with the same error message. Just one note, I also followed the instruction for the DPUCAHX8L version, everything looks fine, all the issue I mentioned here and above are from the DPUCAHX8H version.

wanghy-xlnx, 2021-05-12:
Hi @weberxzq ,
We are sorry to find that the densebox model has some issues  on DPUCAHX8L, and will try to solve it in Vitis AI 1.4. The new model I privided to you still can’t solve the problem, so I also deleted that new model. Please change to other samples or models on the DPUCAHX8L, or replace a DPU IP. 
It should be noted that the compilation of the model corresponds to the DPU IP. If you change the DPU IP, then also use the model compiled for the corresponding DPU IP.
Please refer to https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1354-xilinx-ai-sdk.pdf

weberxzq, 2021-05-12:
Hi @wanghy-xlnx,
Thank you so much for the continuous help. I really appreciate. OK, at this moment I will work on the H version though the low latency L version will be the ideal version fits my need. Again thank you for the advice.

closed 2021-05-12 by weberxzq