Xilinx/Vitis-AI #406: Conv1D support for DPU

YacobHampelt, 2021-05-06:
Hey,

I would like to run temporal convolutional models for time series predictions on the DPU which use 1D convolutions. The models are trained in keras/tensorflow. Sadly I found out that Conv1D operations are not supported by the vitis ai compiler and DPU. I am a bit surprised, as I thought 1D convolutions should be even easier to be implemented. 
Is there any reason why 1D convolutions are not supported and is there any plan to support them in the future?

Thanks

YacobHampelt, 2021-05-10:
Hey, thanks for your reply.
I was thinking about this aswell, but theres one problem. Temporal convolutional models require "causal padding" for the convolution operation. Sadly, keras does not support causal padding for Conv2D operations, only for Conv1D. TensorFlow itself might be able to do causal padding for Conv2D as well, when explicit padding is used. But in that case, I only can use TensorFlow 1, not TensorFlow 2, as the vitis_ai quantizer for TensorFlow 2 only accepts keras models for quantization, no plain TensorFlow graphs, as far as I have seen. So for TensorFlow 2, keras needs to be used, which does not support causal padding for Conv2D.

qianglin-xlnx, 2021-05-12:
Hi @YacobHampelt 
For Conv1D "causal padding", we can use the following Conv2D "valid" mode to replace it equivalently. And we have tested it successfully with Vitis-AI. Please refer to the following code. 

`import numpy as np
import tensorflow as tf
SEQ_LEN = 10
IN_CHANNLES = 12
OUT_CHANNLES = 14
KERNEL_SIZE = 2
DILATION = 8
weights = np.random.randn(KERNEL_SIZE, IN_CHANNLES, OUT_CHANNLES)
bias = np.random.randn(OUT_CHANNLES)
weights_initializer = tf.keras.initializers.Constant(weights)
bias_initializer = tf.keras.initializers.Constant(bias)
x = tf.random.normal((1,SEQ_LEN,IN_CHANNLES))
# conv1d causal
l1 = tf.keras.layers.Conv1D(OUT_CHANNLES, KERNEL_SIZE, dilation_rate=DILATION, padding='causal', kernel_initializer=weights_initializer, bias_initializer=bias_initializer)
res1 = l1(x)
# conv1d valid
l2 = tf.keras.layers.Conv1D(OUT_CHANNLES, KERNEL_SIZE, dilation_rate=DILATION ,padding='valid', kernel_initializer=weights_initializer, bias_initializer=bias_initializer)
pattern = [[0, 0], [DILATION * (KERNEL_SIZE - 1), 0], [0, 0]]
x2 = tf.pad(x, pattern)
res2 = l2(x2)
# conv2d valid
l3 = tf.keras.layers.Conv2D(OUT_CHANNLES, (KERNEL_SIZE, 1), dilation_rate=DILATION ,padding='valid', kernel_initializer=weights_initializer, bias_initializer=bias_initializer)
x3 = tf.reshape(x, (x.shape[0], x.shape[1], 1, x.shape[2]))
pattern = [[0, 0], [DILATION * (KERNEL_SIZE - 1), 0], [0, 0], [0, 0]]
x3 = tf.pad(x3, pattern)
res3 = l3(x3)
res3 = tf.reshape(res3, (res3.shape[0], res3.shape[1], res3.shape[3]))
print(f"res1==res2: {tf.math.equal(res1, res2).numpy().all()}")
print(f"res1==res3: {tf.math.equal(res1, res3).numpy().all()}")`

YacobHampelt, 2021-05-12:
Hey,

thank you SO much! You just helped me a lot! I just tried it myself and can confirm it works perfectly and compiles with vai_c_tensorflow. 
I even tried the tf.pad yesterday, but only for Tensorflow2. Compiling that to an xmodel with vai_c_tensorflow2 didn't work (some error regarding padding), so I gave it up. But after your post I figured maybe it works with Tensorflow 1 and indeed, it does.

Thank you a lot for your effort and your very clear example! I appreciate that a lot.

Have a nice day.

Best regards

closed 2021-05-12 by YacobHampelt