Xilinx/Vitis-AI #409: U250 Vitis AI library error!

zy486at189cn, 2021-05-09:
I use U250 felllow https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/README.md#running-vitis-ai-library-examples-on-alveo-u200alveo-u250-with-dpucadx8g
an error encounter:

(vitis-ai-caffe) Vitis-AI /vitis_ai_home/tools/Vitis-AI-Library >  ~/build/build.Ubuntu.18.04.x86_64.Release/Vitis-AI-Library/classification/test_classification inception_v1 person.jpg
[libprotobuf FATAL google/protobuf/stubs/common.cc:68] This program requires version 3.11.0 of the Protocol Buffer runtime library, but the installed version is 3.4.0.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "google/protobuf/any.pb.cc".)
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  This program requires version 3.11.0 of the Protocol Buffer runtime library, but the installed version is 3.4.0.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "google/protobuf/any.pb.cc".)
Aborted (core dumped)

zy486at189cn, 2021-05-09:
why?

hchenji, 2021-06-01:
same issue here

sumitn-xilinx, 2021-06-02:
Trying to reproduce.

AnushaPerla, 2021-06-02:
Hi @zy486at189cn 

You might have missed the step exporting LD_LIBRARY_PATH. Could you please try exporting LD_LIBRARY_PATH before running.
`export LD_LIBRARY_PATH=~/.local/Ubuntu.18.04.x86_64.Release/lib/:$LD_LIBRARY_PATH`

hchenji, 2021-06-02:
I did do that. When I do an ldd on the classification binary, I see that it is linked to two different libprotobufs: one in usr local lib (v3.4), and one from the caffe environment (v3.11).

originforce, 2021-06-03:
i got exactly the same issue

originforce, 2021-06-03:
i got exactly the same issue while running yoloV3 example

closed 2022-03-03 by qianglin-xlnx