Xilinx/Vitis-AI #410: vivado DPU-TRD flow route_design ERROR

arrow53, 2021-05-13:
Following these instructions

https://github.com/Xilinx/Vitis-AI/blob/1.3.2/dsa/DPU-TRD/prj/Vivado/README.md

I'm getting a route error. 

I have two critical warnings and I'm not sure what they mean

 [Designutils 20-1280] Could not find module 'top_rst_gen_clk_dsp_0'. The XDC file /home/rcagley/Vitis-AI-1.3.2/dsa/DPU-TRD/prj/Vivado/srcs/top/ip/top_rst_gen_clk_dsp_0/top_rst_gen_clk_dsp_0_board.xdc will not be read for any cell of this module.

 [Designutils 20-1280] Could not find module 'top_rst_gen_clk_dsp_0'. The XDC file /home/rcagley/Vitis-AI-1.3.2/dsa/DPU-TRD/prj/Vivado/srcs/top/ip/top_rst_gen_clk_dsp_0/top_rst_gen_clk_dsp_0.xdc will not be read for any cell of this module.

davidxudong, 2021-05-17:
Hi @arrow53 

I think you have deleted the module 'top_rst_gen_clk_dsp_0' or changed it's name.
If you have done that, you have to modify the xdc file accordingly.

Thanks.

arrow53, 2021-05-17:
@davidxudong thanks for the reply. This is an unmodified project from 1.3.2. 

The issue appears to be instantiating more than 1 core in the dpu. If I re-generate the project with one dpu it builds fine
```
dict set dict_prj dict_param  DPU_NUM                {1}
```
I'm not really sure why it is manifesting with this error.

arrow53, 2021-05-17:
ok, if I change the default name to match what it wants with 2 DPUs then it also works. I can't really figure out what is happening but at least it builds so I'll just close.

closed 2021-05-17 by arrow53