Xilinx/Vitis-AI #413: Problem with Dilations

YacobHampelt, 2021-05-17:
Hey,

I am trying to create a model with keras/tensorflow that uses dilations and compile it for DPU. The model is used for time series classification, thus the kernel and dilation have a height = 1 and width > 1 (3 in this case).
Quantization with vai_q_tensorflow works well, however compiling of the quantized model with vai_c_tensorflow yields the following error:

[ERROR] The last dimention of 'matrx1' should be equal to the first dimension of 'matrix2' : matrix1: {matrix1.shape}, matrix2: {matrix2.shape}"
AssertionError: [ERROR] The last dimention of 'matrx1' should be equal to the first dimension of 'matrix2' : matrix1: [1, 3200], matrix2: [3072, 2]

I have created a minimal training script to reproduce the example:

```

from tensorflow.keras.layers import Dense, Conv2D, Activation, Flatten, Input
from tensorflow.keras.models import Sequential, save_model
import tensorflow as tf
import numpy as np

# Generate some random data
data = np.random.normal(size=(1000,1,100,10))
data_y = np.random.randint(low = 0, high = 1, size=(1000))

data_test = np.random.normal(size=(1000,1,100,10))
data_test_y = np.random.randint(low = 0, high = 1, size=(1000))

# Define model
input = Input(shape=(1,100,10), name="input")

conv = Conv2D(32, (1,3), padding='valid', dilation_rate=(1,2))(input)

flatten = Flatten()(conv)
dense = Dense(2, activation = 'softmax')(flatten)

model = tf.keras.Model(inputs = input, outputs=dense)

# Train
loss = tf.keras.losses.SparseCategoricalCrossentropy()
model.compile(optimizer="Adam", loss=loss, metrics=['accuracy'])

model.fit(data, data_y, batch_size=100,
 epochs=5, validation_data=(data_test, data_test_y))

# Save frozen graph

print ("Saving")
graph = tf.get_default_graph()
input_graph_def = graph.as_graph_def()
sess = tf.Session()
init = tf.initialize_all_variables()
sess.run(init)
output_name = "dense/BiasAdd"
output_graph_def = tf.graph_util.convert_variables_to_constants(
                        sess, input_graph_def, output_name.split(","))  

output_graph= "test_model.pb"
with tf.gfile.GFile(output_graph, "wb") as f:
    f.write(output_graph_def.SerializeToString())

```

After training, I save the model to a protobuf / frozen graph as seen in the code above. Afterwards, I run:

```
vai_q_tensorflow quantize --input_frozen_graph test_model.pb \
                          --input_nodes input \
                          --output_nodes dense/BiasAdd \
                          --input_shapes ?,1,100,10 \
                          --input_fn input_fn_test.calib_input

```
Wheres input_fn_test.py is as follows:

```
import numpy as np
def calib_input(iter):
    
  data = np.random.normal(size=(1,1,100,10))

  return {"input": data}

```

With the quantized model, I then run:

`vai_c_tensorflow -f quantize_results/quantize_eval_model.pb -a DPU/test.json -o XModelOutput -n Test`

It seems that after quantization, the Conv2D layer with dilation has a different output matrix size than during training. The error is yield during the dense layer, as the Flatten layer can work with different matrix sizes. However, if I replace the flatten layer by a tf.reshape layer, I get a similiar error like "cannot reshape array of size 3200 to array of size 3072". So the problem definitely lies within the Conv2D layer and it only happens when I use a dilation > 1.

Am I doing something wrong here? As far as I have seen from the documents, dilation is supported by Vitis tools. Is this a bug?
I inspected the graph with netron and figured that before the Conv2D theres a SpaceToBatchND layer and afterwards a BatchSpaceToND layer automatically added by TensorFlow/keras. Might this be a problem? 

I am using the most recent version of Vitis AI, tag 1.3.2 of the github repo.

Thanks for your help!

Best regards

zhenzhen-AMD, 2022-07-22:
Hi @YacobHampelt  ，
We will give you a patch later to fix it. thanks.

zhenzhen-AMD, 2022-07-22:
Hi @YacobHampelt ,
"vai_c_tensorflow -f quantize_results/quantize_eval_model.pb -a **DPU/test.json** -o XModelOutput -n Test"
Can you provide the "DPU/test.json" _file? Thanks.

zhenzhen-AMD, 2022-07-29:
hi @YacobHampelt ,
The patch has been generated. Please use the following command to update:
wget -O xnnc-2.5.0-py36hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py36hd51751d_201.tar.bz2
wget -O xnnc-2.5.0-py37hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py37hd51751d_201.tar.bz2
inside docker. : 
conda update -n vitis-ai-tensorflow xnnc-2.5.0-py36hd51751d_201.tar.bz2
conda update -n vitis-ai-tensorflow2 xnnc-2.5.0-py37hd51751d_201.tar.bz2

qianglin-xlnx, 2022-08-31:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-08-31 by qianglin-xlnx