Xilinx/Vitis-AI #416: Errors encountered when compile custom model using caffe xilinx framework

lebacuong, 2021-05-21:
Hi all,

I am trying to deploy my simple model to ZCU104 BOARD using the caffe framework. After i had quantized my float model, i tried to compile my deploy model to the xmodel using vai_c_caffe command. However, there are some errors encountered according to the local variable referenced before assignment.I tried to fix the error with my knowledge but it didn't work so I want to ask you how to solve this. My float model, deploy model and errors I have attached are in the following [link.](https://drive.google.com/file/d/1b6qjAXtLYDO3d_WBP1uRjwHsn29B8Q-Y/view?usp=sharing)
Thanks you !

lebacuong, 2021-05-21:
Hi all,

I am trying to deploy my simple model to ZCU104 BOARD using the caffe framework. After i had quantized my float model, i tried to compile my deploy model to the xmodel using vai_c_caffe command. However, there are some errors encountered according to the local variable referenced before assignment.I tried to fix the error with my knowledge but it didn't work so I want to ask you how to solve this. My float model, deploy model and errors I have attached are in the this [link.](https://drive.google.com/file/d/1CW4iMRXUKqKV5dcnOv7MFFxKwdb9BN5W/view?usp=sharing)
Thanks you !

qianglin-xlnx, 2021-05-28:
Hi @lebacuong 
Thank you for your feedback. It's a bug of compiler in VAI1.3 and the bug has been fixed in VAI1.4. Stay tuned for VAI1.4.

qianglin-xlnx, 2022-03-02:
Hi @lebacuong 
The bug has been fixed in VAI1.4.

closed 2022-03-02 by qianglin-xlnx