Xilinx/Vitis-AI #417: How to generate the xmodel file for a custom model which is not listed in the model-zoo?

JianmingTONG, 2021-05-22:
Hi 

I've designed my own DNN model and want to deploy the model on the Ultra96, ZCU104 and Alevo U280. Might I ask how could I use the vitis AI framework to generate the xmodel files of my own DNN model?

Thanks in advance.
JT

qianglin-xlnx, 2021-05-27:
Hi @JianmingTONG 
You can refer to the following flow. 
float model -> quantization -> compilation -> deployment.
For details, please refer to https://www.xilinx.com/html_docs/vitis_ai/1_3/zmw1606771874842.html

brisker, 2021-07-01:
@qianglin-xlnx 
1. how to generate the dpu.bit, dpu.hwh,dpu.xclbin file here: https://lpcv.ai/2021LPCVC/fpga-track  ??
2. if there is some custom operations in a pytorch model, such as roi-align, will the  tutorials [here](https://www.xilinx.com/html_docs/vitis_ai/1_3/zmw1606771874842.html) can also generate the right xmodel?  what about the custom operations?

wanghong4compiler, 2022-03-02:
Hi @brisker ,
vitis-ai-1.4 starts to support the custom operations. it could generate the correct xmodel, but the custom operations would be assign to execute on CPU.