Xilinx/Vitis-AI #419: vai_q_caffe: libcaffe-nv.so.0.15: undefined symbol: _ZN6google21kLogSiteUninitializedE

xelalin, 2021-05-23:
Hi 
I try to rebuild vitis-ai-gpu dokcer image by setup/docker_build_gpu.sh and it can success to buildup.

when I use vai_q_caffe to check it and found below error.

(vitis-ai-caffe) Vitis-AI /workspace > vai_q_caffe 
vai_q_caffe: symbol lookup error: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/../lib/libcaffe-nv.so.0.15: undefined symbol: _ZN6google21kLogSiteUninitializedE

how to fix it? 

Thanks,

Alex

xelalin, 2021-05-25:
No matter to build GPU/CPU docker from recipe, there are the same problem as below

```
==========================================
 
__      ___ _   _                   _____
\ \    / (_) | (_)            /\   |_   _|
 \ \  / / _| |_ _ ___ ______ /  \    | |
  \ \/ / | | __| / __|______/ /\ \   | |
   \  /  | | |_| \__ \     / ____ \ _| |_
    \/   |_|\__|_|___/    /_/    \_\_____|
 
==========================================

Docker Image Version:  latest 
Build Date: 2021-05-25
VAI_ROOT: /opt/vitis_ai

For TensorFlow 1.15 Workflows do:
     conda activate vitis-ai-tensorflow 
For Caffe Workflows do:
     conda activate vitis-ai-caffe 
For Neptune Workflows do:
     conda activate vitis-ai-neptune 
For PyTorch Workflows do:
     conda activate vitis-ai-pytorch 
For TensorFlow 2.3 Workflows do:
     conda activate vitis-ai-tensorflow2 
Vitis-AI /workspace > conda activate vitis-ai-caffe 
(vitis-ai-caffe) Vitis-AI /workspace > vai_q_caffe --help
vai_q_caffe: symbol lookup error: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/../lib/libcaffe-nv.so.0.15: undefined symbol: _ZN6google10LogMessageC1EPKciiiMS0_FvvE
(vitis-ai-caffe) Vitis-AI /workspace > cd /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/../lib
(vitis-ai-caffe) Vitis-AI /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib > ls -la libcaffe-nv.so.0.15
lrwxrwxrwx 1 1000 1000 22 May 24 23:07 libcaffe-nv.so.0.15 -> libcaffe-nv.so.0.15.14
(vitis-ai-caffe) Vitis-AI /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib > ls -la libcaffe-nv.so*
lrwxrwxrwx 1 1000 1000      22 May 24 23:07 libcaffe-nv.so -> libcaffe-nv.so.0.15.14
lrwxrwxrwx 1 1000 1000      22 May 24 23:07 libcaffe-nv.so.0.15 -> libcaffe-nv.so.0.15.14
-rw-rw-r-- 1 1000 1000 9060392 May 24 23:07 libcaffe-nv.so.0.15.14
(vitis-ai-caffe) Vitis-AI /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib > ldd -r libcaffe-nv.so.0.15
	linux-vdso.so.1 (0x00007ffd415a2000)
	libcrypto.so.1.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libcrypto.so.1.1 (0x00007feec5f0c000)
	libboost_system.so.1.65.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libboost_system.so.1.65.1 (0x00007feec5d07000)
	libboost_thread.so.1.65.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libboost_thread.so.1.65.1 (0x00007feec5ae4000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007feec58c5000)
	libboost_filesystem.so.1.65.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libboost_filesystem.so.1.65.1 (0x00007feec56a9000)
	libboost_regex.so.1.65.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libboost_regex.so.1.65.1 (0x00007feec53a7000)
	libglog.so.0 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libglog.so.0 (0x00007feec6adc000)
	libgflags.so.2.2 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libgflags.so.2.2 (0x00007feec6ab8000)
	libprotobuf.so.22 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libprotobuf.so.22 (0x00007feec510e000)
	libhdf5.so.103 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libhdf5.so.103 (0x00007feec4d78000)
	libhdf5_hl.so.100 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libhdf5_hl.so.100 (0x00007feec6a92000)
	libyaml-cpp.so.0.6 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libyaml-cpp.so.0.6 (0x00007feec6a16000)
	liblmdb.so => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././liblmdb.so (0x00007feec69ff000)
	libleveldb.so.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libleveldb.so.1 (0x00007feec4b19000)
	libopencv_highgui.so.3.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopencv_highgui.so.3.4 (0x00007feec69b7000)
	libopenblas.so.0 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopenblas.so.0 (0x00007feec2f40000)
	libpython3.6m.so.1.0 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libpython3.6m.so.1.0 (0x00007feec2bf9000)
	libboost_python3.so.1.65.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libboost_python3.so.1.65.1 (0x00007feec29bb000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007feec261d000)
	libopencv_videoio.so.3.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopencv_videoio.so.3.4 (0x00007feec6978000)
	libopencv_imgcodecs.so.3.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopencv_imgcodecs.so.3.4 (0x00007feec1e7a000)
	libopencv_imgproc.so.3.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopencv_imgproc.so.3.4 (0x00007feebf2cc000)
	libopencv_core.so.3.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libopencv_core.so.3.4 (0x00007feebe202000)
	libstdc++.so.6 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libstdc++.so.6 (0x00007feebe08e000)
	libgcc_s.so.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././libgcc_s.so.1 (0x00007feec6962000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007feebdc9d000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007feebda99000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007feebd891000)
	/lib64/ld-linux-x86-64.so.2 (0x00007feec6915000)
	libicudata.so.58 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././libicudata.so.58 (0x00007feebbf8f000)
[... 135 lines cut ...]
	libjasper.so.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././libjasper.so.1 (0x00007f9e21a7e000)
	libgomp.so.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././libgomp.so.1 (0x00007f9e21a4f000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f9e217c3000)
	libgthread-2.0.so.0 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././././libgthread-2.0.so.0 (0x00007f9e217be000)
[... cut ...]
	libhogweed.so.4 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././././libhogweed.so.4 (0x00007f9e20855000)
	libgmp.so.10 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././././libgmp.so.10 (0x00007f9e207af000)
	libX11.so.6 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/./././libX11.so.6 (0x00007f9e2066a000)
	libxcb.so.1 => /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/././././libxcb.so.1 (0x00007f9e2063e000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f9e2043a000)
(vitis-ai-caffe) Vitis-AI /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib > 
```

zy486at189cn, 2021-06-13:
meet the same problem

xelalin, 2021-06-16:
I fixed the issue, The cause is that the glog conda package had been updated from 0.4.0 to 0.5.0 at conda-forge.
please use below command to fix it.
> conda activate < conda_name >
> sudo env PATH=/opt/ vitis_ai conda /bin:$PATH CONDA_PREFIX=/opt/vitis_ai/conda/envs/< conda_name > conda install --force --yes glog=0.4.0

zy486at189cn, 2021-06-18:
@xelalin  nice good job!

zy486at189cn, 2021-06-18:
conda activate vitis-ai-caffe 
sudo env PATH=/opt/vitis_​​ai/conda/bin:$PATH CONDA_PREFIX=/opt/vitis_ai/conda/envs/vitis-ai-caffe/ conda install --force --yes glog=0.4.0

zy486at189cn, 2021-06-19:
@xelalin 
but vai_q_caffe error:

vai_q_caffe quantize -model ResNet_50_train_val.prototxt -calib_iter 100 -weights resnet_iter_180000.caffemodel -output_dir results/ -method 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0619 05:16:37.151515 14611 gpu_memory.cpp:53] GPUMemory::Manager initialized with Caching (CUB) GPU Allocator
I0619 05:16:37.151587 14611 gpu_memory.cpp:55] Total memory: 11720130560, Free: 11127160832, dev_info[0]: total=11720130560 free=11127160832
I0619 05:16:37.152267 14611 vai_q.cpp:260] Using GPUs 0
I0619 05:16:37.152660 14611 vai_q.cpp:265] GPU 0: GeForce GTX 1080 Ti
F0619 05:16:38.533720 14611 convert_proto.cpp:2618] Check failed: b_size == 3 Wrong Blobs for BatchNorm Layer bn_conv1, BatchNorm with 5 blobs should not followed by Scale layer.
*** Check failure stack trace: ***
Aborted (core dumped)

sand-is-fur, 2021-08-06:
@xelalin 
I ran the command: sudo env PATH=/opt/vitis_​​ai/conda/bin:$PATH CONDA_PREFIX=/opt/vitis_ai/conda/envs/vitis-ai-caffe/ conda install --force --yes glog=0.4.0

but I get an error _**EnvironmentNotWritableError: The current user does not have write permissions to the target environment.
  environment location: /opt/vitis_ai/conda/envs/vitis-ai-optimizer_caffe/**_

Do you know if there is a way past this issue?

zy486at189cn, 2021-08-06:
just use sudo conda install.....

closed 2022-02-22 by niuxjxlnx

yydc-0, 2022-07-22:
> I fixed the issue, The cause is that the glog conda package had been updated from 0.4.0 to 0.5.0 at conda-forge. please use below command to fix it.

nice job!!!