Xilinx/Vitis-AI #420: VART for resnet50 demo error : Aborted

ctlinglingliu, 2021-05-25:
when i run the demo resnet50 of VART on the board , it has a mistake below:
./resnet50 resnet50.elf 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1127 01:02:05.563463   864 main.cc:286] create running for subgraph: resnet50_0
Aborted
 i don't know the reason ,and it has no detail log of the error

ctlinglingliu, 2021-05-26:
hi @qianglin-xlnx
I use the version 1.2 of Vitis-AI and flow the step in the readme that support elf model

levent9402, 2022-02-17:
Hi， I have not been able to reproduce the problem you described. Could you please try the latest version?
https://github.com/Xilinx/Vitis-AI/tree/master/demo/VART/resnet50

guohaot-Xlnx, 2022-02-23:
Hi @ctlinglingliu 
This issue is about to close. If you have any questions, please repost to the relevant section

closed 2022-02-23 by guohaot-Xlnx