Xilinx/Vitis-AI #421: DPU.get_input_tensors

dingguodong-826, 2021-05-25:
Hello, you have done a good job. I have been using the AXU4EV development board to deploy the network recently. After completing the quantization and compilation of the model, I tested my model on the development board by referring to the routine of handwritten number recognition, but I met some problems:
When I was quantifying the network, the input of the given network quantization in the quanter.py file was in the format of BCHW, as shown in the figure:
[image]
57/5000 
However, when I run on the development board, under DPU.GET_INPUT_TENSORS, I print the dimension information and it shows BHWC instead.
[image]
It should be noted that not only the input, but also the output is in BHWC format, I don't know why this is the case

qianglin-xlnx, 2021-05-25:
@dingguodong-826 
DPU only support NHWC format. When the input is NCHW, it will be automatically converted to NHWC by VART and sent to DPU.

dingguodong-826, 2021-05-26:
> @dingguodong-826
> DPU only support NHWC format. When the input is NCHW, it will be automatically converted to NHWC by VART and sent to DPU.

Thank you for your reply！It is for this reason, I found a picture into the network, after the output on my computer and I on the development board using the compiled model output does not agree, I think that is the loss of the quantitative part accuracy is too large, if say to you, only support BHWC DPU, so if I don't have to pay attention to the computer network and development board on the output of the inconsistent problem. If the DPU only supports BHWC, will it affect the final performance of the network deployed on the development board?

levent9402, 2022-02-17:
Normally, it does not affect final performance or accuracy. However, it is possible to change the order of post-processing operations, which degrades cache performance and results in final performance deterioration. For example, the wrong order of memory operations can result in more frequent cache misses

> Thank you for your reply！It is for this reason, I found a picture into the network, after the output on my computer and I on the development board using the compiled model output does not agree, I [...]

guohaot-Xlnx, 2022-02-23:
This issue is about to close. If you have any questions, please repost to the relevant section

closed 2022-02-23 by guohaot-Xlnx

AfifaIshtiaq, 2022-06-01:
> @dingguodong-826 DPU only support NHWC format. When the input is NCHW, it will be automatically converted to NHWC by VART and sent to DPU.

Hallo, I currently face issue with the following : I have attached my app script please  let me know whats the issue
[app_deeprx.zip](https://github.com/Xilinx/Vitis-AI/files/8813710/app_deeprx.zip)

root@xilinx-vck190-2021_2:~/deeprx_app# python3 app_deeprx.py
(14, 72, 1)
Pre-processing 10800 images...
Found 1 DPU subgraphs
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
Starting 1 threads...
Starprocess
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "app_deeprx.py", line 80, in runThread
    deeprx_dict, deeprx_inbuffer, deeprx_outbuffer = init_dpu_runner(deeprx_dpu_runner)
  File "app_deeprx.py", line 54, in init_dpu_runner
    dpu_inputs = dpu_runner.get_input_tensors()
AttributeError: 'list' object has no attribute 'get_input_tensors'
------------------------------------
Throughput=4006587.94 fps, total frames = 10800, time=0.0027 seconds