Xilinx/Vitis-AI #422: Custom pointpillars model on ZCU104

reddittoob, 2021-05-26:
Hello, I'm trying to run pointpillars model on ZCU104.

I got a well result on my Desktop using my custom dataset like below.

[image]

However, on ZCU104, results is not good and any object cannot be detected. Quantization & Compile is no problem. I thought this problem is occurred because of Calibration data in ZCU104 application. 

Can I remove a calibration on board?

Please let me know about it.

Thank you.

lishixlnx, 2021-05-26:
can you get correct detect result when running the Quantization model with python code?

reddittoob, 2021-05-26:
Can I run python code using xmodel?

If I run python code, I got an error like

Traceback (most recent call last):
  File "code/test/image_demo.py", line 380, in <module>
    dt_annos = evaluate_single(input_dict, args.config, args.weights, args.quant_mode, args.quant_dir)
  File "code/test/image_demo.py", line 290, in evaluate_single
    restore(ckpt_path, net)
  File "code/test/image_demo.py", line 261, in restore
    ckpt_dict = torch.load(ckpt_path)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/serialization.py", line 529, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/serialization.py", line 692, in _legacy_load
    magic_number = pickle_module.load(f, **pickle_load_args)
_pickle.UnpicklingError: invalid load key, '\x0a'.

Do you have solution?

Treemann, 2021-05-26:
Before dumping the xmodel, is the accuracy after Calibration&Quantization good enough compared with that of your float model?

reddittoob, 2021-05-27:
After calibration & quantization, I can't run model because of the error mentioned above. 
So, I can't compare float model.

Treemann, 2021-05-27:
Yeah, the .xmodel is for DPU deployment and could not be loaded and run in the python code.
I mean after you train your float model, you must have run 'code/test/run_quant.sh' to do calibration, quantization and dumping xmodel. And the following command in 'code/test/run_quant.sh' should give a accuracy about the quantized model. You should compare the quantized accuracy with your float accuracy to see if there is any problem at quantization. 
```bash
echo "Testing quantized model..."
QUANT_MODE=test
python code/test/test.py -config ${CONFIG} -weights ${WEIGHTS} -pred_dir ${RESULT} -quant_mode ${QUANT_MODE} -quant_dir ${QUANT_DIR}
```

reddittoob, 2021-05-28:
Yes, I already run command you told. Accuracy is nearly same.
So, I thought quantization and calibration is no problem.

lishixlnx, 2021-05-28:
where do you get the test data?
where is the calibration data from?
please note that you may need changing the data in get_display_data() in your test code.

reddittoob, 2021-05-28:
I have our custom data.
Also, because I don't have calibration data, I have changed code that can train using only lidar data.

lishixlnx, 2022-02-23:
hi reddittoob, what is the status of this issue? can it be closed now?

qianglin-xlnx, 2022-03-01:
Hi @reddittoob 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-01 by qianglin-xlnx