Xilinx/Vitis-AI #424: meet error when cross compiling resnet50

JunnanLi, 2021-05-27:
when I follow the command of setting up cross-compiler, it has a mistake below (bash -x build.sh):
~/Vitis-AI-master/demo/VART/resnet50/../common/common.h:34:10: fatal error: xir/util/data_type.hpp: No such file or directory
   34 | #include <xir/util/data_type.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
i don't know the reason, can you help me?

qianglin-xlnx, 2021-05-28:
Hi @JunnanLi 
Could you find the data_type.hpp file under your cross-compiler installation path? If the cross-compiler is installed correctly, you will find the file, as shown below.
xxxxxxxxxxxxx:petalinux_sdk% find ./ -name data_type.hpp
./sysroots/aarch64-xilinx-linux/usr/include/xir/util/data_type.hpp

If not, please check the step 4 in https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART#step1-setup-cross-compiler
Make sure  vitis_ai_2020.2-r1.3.2.tar.gz is installed correctly.

JunnanLi, 2021-05-28:
I have run the command "tar -xzvf vitis_ai_2020.2-r1.3.2.tar.gz -C ~/petalinux_sdk/sysroots/aarch64-xilinx-linux", and it has only two files under ~/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/include/xir/util:
xxxxxxxx:~/petalinux_sdk/sysroots/aarch64-xilinx-linux$ ls usr/include/xir/util/
any.hpp  tool_function.hpp

closed 2021-05-28 by JunnanLi