Xilinx/Vitis-AI #425: U280 Vitis AI

karan18205, 2021-05-28:
I am trying to run Demo on Vitis AI and getting following error:

[XRT] ERROR: See dmesg log for details. err=-33
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0528 02:43:20.415526   785 xrt_bin_stream.cpp:149] Check failed: xclLoadXclBin(handle, blob) == 0 (-33 vs. 0) Bitstream download failed !
*** Check failure stack trace: ***
Aborted (core dumped)

Also when I see dmesg I get the following output:

[  +0.000486] xocl 0000:84:00.1: _xocl_drvinst_open: OPEN 1
[  +0.000006] [drm] creating scheduler client for pid(32707), ret: 0
[  +0.015009] mailbox.u mailbox.u.13631488: mailbox_request: sending request: 9 via HW
[  +0.000458] mailbox.m mailbox.m.13631488: process_request: received request from peer: 9, passed on
[  +0.000007] icap.m icap.m.15728640: icap_ocl_update_clock_freq_topology: Num clocks is 3
[  +0.000004] icap.m icap.m.15728640: icap_ocl_update_clock_freq_topology: requested frequency is : 32733 xclbin freq is: 300
[  +0.000003] icap.m icap.m.15728640: icap_ocl_update_clock_freq_topology: Unable to set frequency as requested frequency 32733 is greater than set by xclbin 300
[  +0.001686] mailbox.m mailbox.m.13631488: mailbox_post_response: posting response for: 9 via HW
[  +0.000140] xocl 0000:84:00.1: xocl_reclock_ioctl: xocl_reclock_ioctl err: -33
[  +0.045146] [drm] client exits pid(32707)
[  +0.000015] xocl 0000:84:00.1: xocl_drvinst_close: CLOSE 2
[  +0.000005] xocl 0000:84:00.1: xocl_drvinst_close: NOTIFY ffff9858f0f60810

Does anyone having this problem?

berankilic, 2021-05-28:
Did you try xbutil reset?

karan18205, 2021-05-28:
Yes I did, now the output of dmesg is:

 +0.019857] [drm] client exits pid(33002)
[  +0.000006] xocl 0000:84:00.1: xocl_drvinst_close: CLOSE 2
[  +0.000003] xocl 0000:84:00.1: xocl_drvinst_close: NOTIFY ffff9858f0f60810
[  +0.000355] xocl 0000:84:00.1: _xocl_drvinst_open: OPEN 1
[  +0.000003] [drm] creating scheduler client for pid(33002), ret: 0
[  +0.021063] mailbox.u mailbox.u.13631488: mailbox_request: sending request: 9 via HW
[  +0.000259] mailbox.m mailbox.m.13631488: process_request: received request from peer: 9, passed on
[  +0.000010] icap.m icap.m.15728640: icap_ocl_update_clock_freq_topology: ERROR: **There isn't a hardware accelerator loaded in the dynamic region. Validation of accelerator frequencies cannot be determine**
[  +0.005675] mailbox.m mailbox.m.13631488: mailbox_post_response: posting response for: 9 via HW
[  +0.000570] xocl 0000:84:00.1: xocl_reclock_ioctl: xocl_reclock_ioctl err: -33
[  +0.045632] [drm] client exits pid(33002)
[  +0.000011] xocl 0000:84:00.1: xocl_drvinst_close: CLOSE 2
[  +0.000004] xocl 0000:84:00.1: xocl_drvinst_close: NOTIFY ffff9858f0f60810

closed 2021-06-17 by karan18205