Xilinx/Vitis-AI #427: How to deal with multiple subgraphs.

vtorres621, 2021-05-28:
Hello,

I have followed Pytorch mnist tutorial succesfully to accelerate a custom network using Vitis AI flow.  Now I am training and trying to deploy my own models but I found the issue that Vitis AI Compiler generates multiple subgraphs.

When my network has the following architecture: Conv-Maxpool -> Flatten -> Dense, VAIC creates one subgraph for the convolutional layers and another for the dense layers.    I have tried creating a DPU runner for each and getting the output from DPU runner 1 into the input from DPU runner 2 but this has given incorrect results.  In contrast, if I change my layer architecture to fully convolutional I do not get this problem and I am able to run inference with correct results.

Here I print the input and output tensors of both dpu runners I created: 
[image]

You can see that the "flattening" operation is not being performed.  I have tried different reshaping methods but I am still getting the same results.

Any suggestions on how to proceed?

Thanks!

wilderfield, 2021-05-28:
Victor, as of today, the only way to automatically handle the scenario with
multiple subgraphs, is to use the TVM flow.
Please check out that flow. Alternatively, you have to manually take the
output of the first DPU subgraph, and manually implement the final layer in
CPU code.

vtorres621, 2021-05-28:
Thanks for your answer Bryan.  
So if I understand correctly, the output of the first subgraph is missing the final layer? I would have to implement that final layer on CPU before inputing to the second DPU subgraph? 
Also is there a way to see which operations are compiled into the subgraphs and which are not?   I remember in previous versions the compiler would print what was on the DPU kernel and what was on the CPU kernel.

Thanks!

VishalX, 2021-05-31:
@vtorres621 You can use [xir](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/xir) APIs to get all the info regarding a generated graph. 

[README](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/xir/README.md)

vtorres621, 2021-07-15:
Just to provide a solution in case someone has the same problem:

1.- My model used "flatten" function between conv-max and dense layers.  For some reason VAI Compler did some Transpose and reshape of the output data from subgraph 1 outside the DPU subgraphs.  Once I implemented this in the CPU, I was able to input this into the second DPU subgraph and get the correct output.  
2.- Another way to avoid this is to use "reshape" instead of "flatten" inside your model definition.  This way the compiler only generates 1 DPU subgraph.

To see what the subgraphs are doing you can print the .xmodel in subgraph form using "xir svg <xmodel> <svg>" inside the Vitis AI docker.  This helped me find out what was required to happend between DPU subgraphs.

closed 2021-07-15 by vtorres621