Xilinx/Vitis-AI #428: Input size error on unet

jonassundseth, 2021-05-30:
Hi, 

I have implemented a unet-like architecture on a ZCU104 and it works great when using input resolution which are powers of two (2^n). In a previous issue I was told that the only constraint should be that the input resolution had to be a multiple of 16, so does anyone know what might be wrong? The problem arises when:
```
inputTensors = runner.get_input_tensors()
```
returns ```256x256``` while the input in this case is ```192x192```, so the reshape fails.
```
imageRun[j, ...] = img[(count + j) % n_of_images].reshape(input_ndim[1:])
ValueError: cannot reshape array of size 36864 into shape (256,256,1)
```

yongshengxu-xlnx, 2021-05-31:
The total number of elements on the `img[(count + j) % n_of_images] & input_ndim[1:]` should be the same while applying the reshape, you need to use resize instead of reshape. 
  But according to your description, you should more hope that the input size of the network becomes 192x192. This requires regenerating a new xmodel with the corresponding size. You need to use the corresponding tool to generate a new xmodel whose input size is 192x192.

jonassundseth, 2021-05-31:
Thank you for answering.

Using resize instead of reshape is not possible as you can see from the error below:
```
imageRun[j, ...] = img[(count + j) % n_of_images].resize(input_ndim[1:])
ValueError: cannot resize this array: it does not own its data
```

I tried regenerating by running the full pipeline again, but I still run into the same error as mentioned above.

yongshengxu-xlnx, 2022-02-22:
Hi, @jonassundseth 
To circumvent your issue you can call resize from numpy (not as an attribute of a ndarray) which will detect this issue and copy the data automatically:
`np.resize(img[(count + j) % n_of_images], input_ndim[1:])`
For details, you can find on stackoverflow. But this way will change the size of your input

qianglin-xlnx, 2022-03-02:
Hi @jonassundseth 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-02 by qianglin-xlnx