Xilinx/Vitis-AI #430: Wrong results while executing a finetuned model on the Alveo U250

estherbernard, 2021-05-31:
Hi, 
We have done a finetuning  with a binary output using a keras model (vgg16 whith imagenet weights) and by adding the layers :
-	Flatten (transformed to Reshape)
-	Relu (x2)
-	Softmax
When we do the inference on GPU using tensorflow (frozen_model.pb), it seems ok (good accuracy) but despite the fact that every layers are compatible (according to the logs), the accuracy is only 52% on the alveo U250. 
We have tried the simulation « quantize_evaluation.py » and it seems ok too (good accuracy like the frozen_model.pb). 
After checking every throughputs, it seems that the problem arrives when we do the Reshape on the finetuning (the throughputs of previsous layers are almost the same)

Could you please tell us why the reshape do not seems to work on the alveo and how we could fix it ?

Thank you in advance

chuo1, 2022-07-21:
Can you provide code and files so that we can reproduce the error

renjiangyong, 2022-09-06:
The issue is existed for a long time and is inactive.  I will close it, if you have problem you can reopen it .

closed 2022-09-06 by renjiangyong