Xilinx/Vitis-AI #432: How to increase FPS while running Yolo-v3 example.

daivikbhatia, 2021-06-01:
Hi 
so I have trained a custom yolo-v3 model trained on voc dataset in darknet and then converted it into caffe. I'm using the old vitis version for some reason that gave me a complied elf file. 
I'm using this code mentioned below for deployment in python on pynq v2.5 image on Ultra96 v2 board
https://github.com/Xilinx/Vitis-AI/blob/1.1/demo/DNNDK/tf_yolov3_voc_py/tf_yolov3_voc.py.
Everything is working fine, I just need to know how can I implement multithreading to increase the output FPS as suggested here
https://www.xilinx.com/html_docs/vitis_ai/1_3/running_examples.html#ariaid-title2

zackdilan, 2021-06-14:
Hey!
I think you can follow this example https://github.com/Xilinx/Vitis-AI/blob/1.1/demo/VART/resnet50_mt_py/resnet50.py, try creating different threads for the DPU process, and make sure you are running different threads for i/o image processes, consider this C++ example (https://github.com/Xilinx/Vitis-AI/blob/1.1/demo/VART/video_analysis/src/main.cc). 
Let me know if you need further help :)

daivikbhatia, 2021-06-17:
Hi, thankyou so much, that helped a lot. I can now use multithreading.
But there is another issue I ran into. Now I'm converting a yolo-v4 model into tensorflow. The conversion, quantization and compilation was sucessful. I got the elf file and the kernal info. The kernal name in kernal info is "dpu_yolov4" but as soon as I put this kernal name in n2cube.dpuLoadKernal("dpu_yolov4"), the jupyter kernal of pynq 2.5 image crashes. Can you help me resolve it?
Also I tried same model for caffe with kernal name "yolov4_voc" and everything ran smoothly. 
I don't think the issue is related to elf file. I think it's n2cube library issue as loading kernal name is 1st step that does not require any model file.

yongshengxu-xlnx, 2022-02-22:
Hi, @daivikbhatia 
Has the problem been solved? If not, please refer to:[]( https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Design_Tutorials/07-yolov4-tutorial)
Or you provide the model with the problem, we will reproduce the problem.

yongshengxu-xlnx, 2022-02-22:
Hi, @daivikbhatia
Has the problem been solved? If not, please refer to:[https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Design_Tutorials/07-yolov4-tutorial](https://github.com/Xilinx/Vitis-AI-Tutorials/tree/master/Design_Tutorials/07-yolov4-tutorial)
Or you provide the model with the problem, we will reproduce the problem.

qianglin-xlnx, 2022-03-02:
Hi @daivikbhatia 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-02 by qianglin-xlnx