Xilinx/Vitis-AI #433: RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.

Darshcg, 2021-06-01:
Hi Xilinx Team,

We are working on custom trained Pytorch model Quantization and Compilation. 

Having said that the Quant model is generated, but while evaluating the quant model we are an facing issue. The issue has been pasted below:

```
      File "quantize.py", line 127, in <module>

        run_main()

      File "quantize.py", line 120, in run_main

        quantize(args.dset_dir,args.float_model,args.quant_mode,args.batchsize,args.quant_model, cfg)

      File "quantize.py", line 72, in quantize

        evaluate(quantized_model,cfg)

      File "/workspace/work_3/eval.py", line 56, in evaluate

        class_p, box_p, anchors = net(img)

      File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__

        result = self.forward(*input, **kwargs)

      File "quant_model/Yolact.py", line 352, in forward

        self.output_module_200 = self.module_200(input=self.output_module_198, size=[self.output_module_199,-1,11])

      File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__

        result = self.forward(*input, **kwargs)

      File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/modules/module_template.py", line 129, in forward

        output = getattr(input, self.op_type, None)(*args, **kwargs)

    RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
```

I am using a xilinx/vitis-ai:latest container for Pytorch Quantization.
It would be a great help from your side if you can help us to resolve the issue.

 

Thanks in advance,
Darshan C G

niuxjxlnx, 2021-07-19:
hi, @Darshcg :

We have fixed similar issue in vitis-ai 1.4, which will be released in a few weeks. But to confirm this, can you tell me which model are you working on?

niuxjxlnx, 2022-02-21:
Suppose this problem has been fixed in Vitis AI 1.4. And will close this issue if no more response.

closed 2022-02-25 by niuxjxlnx

dungng27, 2023-04-06:
> 

> hi, @Darshcg :
> 
> We have fixed similar issue in vitis-ai 1.4, which will be released in a few weeks. But to confirm this, can you tell me which model are you working on?

Hi I face similar issue. Is this also fixed in vitis-ai 1.4.1 and if so how to resolve this?
Thanks.